perfanalytics/pose2sim #104: I want to add muscle to upper limbs

smwscth, 2024-05-07:
Hello,
    I update pose2sim recently, I'm so happy to see pose2sim get even better, thanks for your work!
    I want to do CMC by Body25B model, so I try to add muscle to upper limbs. I have a question about the location of the muscle: I found that five locations that need to specify when building a muscle,  but I don't understand what determines the location. I would really appreciate it if you could tell me how to build a muscle properly.
[image]

davidpagnon, 2024-05-07:
Hi, 

I'm glad people find some use to it, and that some are now getting involved!

I'm not an expert at OpenSim muscle modeling, you may get better answers of you post on [their forum](https://simtk.org/plugins/phpBB/indexPhpbb.php?f=91). I think these locations correspond to points the muscle passes through. If you open this model in the GUI and look at this muscle, you will probably find a red line separated in 4 segments (so 5 points). Check the "muscle geometry" at this page for an answer with more details: https://opensimconfluence.atlassian.net/wiki/spaces/OpenSim/pages/53086215/Gait+2392+and+2354+Models

I haven't thoroughly tested it, but I uploaded a model with muscles that you may want to try and use : https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/OpenSim_Setup/Model_Pose2Sim_Body25b_contacts_muscles.osim

smwscth, 2024-05-09:
Thank you! That should be helpful. I'll try adding muscle testing.

closed 2024-05-09 by smwscth