perfanalytics/pose2sim #106: Problem with synchronization/triangulation

Kamu0209, 2024-05-16:
Hello,

I have a Problem with the workstep Synchronization/Triangulation.
1. The Syncro is not working, if I try it for the first Time in a Trial

[image]

[image]

Maybe its because my calibration output is different from the tutorial:

[image]

I´m using EasyMocap to calibrate my cams and I changed toml file to "Convert" and "Easymocap"

2. 

The output for the personAssociation looks like this:

[image]

Afterwards the Triangulation does not work and I don´t know why:(

[image]

Kamu0209, 2024-05-16:
Problem solved!!!

Deleted other Trial/Batch Folder

davidpagnon, 2024-05-16:
I love it when it is solved without my help 😁
Would you mind telling me what went wrong, in case some other users face it?

Kamu0209, 2024-05-17:
I am not sure. I thought i don´t work because the calibration was bad. I ve only deleted other Folder Batch/Trial Folder , so that i only have one folder with the needed structure and of course i ve selected only the trial folder in anaconda

[image]

Kamu0209, 2024-05-17:
The Cause was I have changed the personAssociation Parameters in the config.toml and now it is working (likelihood_threshold_association and reproj_error_threshold_association)

davidpagnon, 2024-05-18:
Nice, thanks your for following up!

closed 2024-05-18 by davidpagnon