perfanalytics/pose2sim #108: Camera Calibration Error

harshitsinghcode, 2024-06-04:
I have the checkboard images stored in the folder captured from different angles(20), but the very next step which is camera calibration, I a'int getting any tool which can create a .toml file for me. And I'm  bit confused right now over the further steps to follow in order to get the inverse kinematics in opensim.

harshitsinghcode, 2024-06-04:
@davidpagnon asap, help needed!

davidpagnon, 2024-06-04:
Hi, I don't understand your problem, have you read carefully this part of the doc?
https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#calculate-from-scratch

harshitsinghcode, 2024-06-04:
i am doing it with my phone's camera, but im not able to get the calibration from opencv method even.

davidpagnon, 2024-06-04:
Sorry I need more context to be able to help you. At which stage to your struggle? Is there an error message?

harshitsinghcode, 2024-06-04:
i have the photos in my folder, and i tried to calibrate but this error shows - 

```

(base) C:\Users\trimble\Desktop\harshit\pose2sim\Pose2Sim>python cal_cam.py
Traceback (most recent call last):
  File "C:\Users\trimble\Desktop\harshit\pose2sim\Pose2Sim\cal_cam.py", line 45, in <module>
    ret, mtx, dist, rvecs, tvecs = cv.calibrateCamera(objpoints, imgpoints, gray.shape[::-1], None, None)
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cv2.error: OpenCV(4.9.0) D:\a\opencv-python\opencv-python\opencv\modules\calib3d\src\calibration.cpp:3752: error: (-215:Assertion failed) nimages > 0 in function 'cv::calibrateCameraRO'
```

from this code - 

```
import numpy as np
import cv2 as cv
import glob
import os

# Set the termination criteria for the calibration process
criteria = (cv.TERM_CRITERIA_EPS + cv.TERM_CRITERIA_MAX_ITER, 30, 0.001)

# Prepare object points and image points lists
objpoints = []  # 3D points in real-world space
imgpoints = []  # 2D points in image plane

# Define the number of inner corners in the chessboard pattern
chessboard_size = (7, 6)  # Number of inner corners (rows, columns)

# Create object points
objp = np.zeros((chessboard_size[0] * chessboard_size[1], 3), np.float32)
objp[:, :2] = np.mgrid[0:chessboard_size[0], 0:chessboard_size[1]].T.reshape(-1, 2)

# Set the path for the folder containing the chessboard images
image_folder_path = r'C:\Users\trimble\Desktop\harshit\openpose\calibration\intrinsics'
images = glob.glob(os.path.join(image_folder_path, '*.jpg'))

for fname in images:
    img = cv.imread(fname)
    gray = cv.cvtColor(img, cv.COLOR_BGR2GRAY)

    # Find chessboard corners
    ret, corners = cv.findChessboardCorners(gray, chessboard_size, None)

    # If corners are found, refine them and add to the object and image points lists
    if ret:
        objpoints.append(objp)
        corners2 = cv.cornerSubPix(gray, corners, (11, 11), (-1, -1), criteria)
        imgpoints.append(corners2)

        # Draw and display the corners
        cv.drawChessboardCorners(img, chessboard_size, corners2, ret)
        cv.imshow('Chessboard Corners', img)
        cv.waitKey(500)

cv.destroyAllWindows()

# Perform camera calibration
ret, mtx, dist, rvecs, tvecs = cv.calibrateCamera(objpoints, imgpoints, gray.shape[::-1], None, None)

# Save camera matrix and distortion coefficients to a .toml file
calibration_data = {
    'camera_matrix': mtx.tolist(),
    'distortion_coefficients': dist.tolist()
}

with open('calibration.toml', 'w') as f:
    for key, value in calibration_data.items():
        f.write(key + ' = ' + str(value) + '\n')

```

davidpagnon, 2024-06-04:
Before your calibrateCamera call, can you do a print(images)?
And after the if ret, can you do a `else: print('no corner found')`?
It seems like it has not found any images.

But where does that code come from? Why don't you use the regular Pose2Sim function?

harshitsinghcode, 2024-06-04:
The function, pose2sim doesn't work on this....I tried working with that first but it shows "calibrate" doesn't exist in pose2sim

harshitsinghcode, 2024-06-05:
@davidpagnon 
[shiv.zip](https://github.com/user-attachments/files/15573310/shiv.zip)
this is my extrinsic and intrinsic file, can u once see through it?

harshitsinghcode, 2024-06-05:
@davidpagnon 
i've some quick doubts, ive been on this repo since a week, and i m close to implementing this. can you please be here to clarify my some of the theoretical doubts on a rapid fire. im stuck. appreciation to your work!

davidpagnon, 2024-06-05:
Sorry I'm quite busy this week :)

Can you first try that?

> Before your calibrateCamera call, can you do a print(images)?
> And after the if ret, can you do a `else: print('no corner found')`?
> It seems like it has not found any images.

davidpagnon, 2024-06-05:
Also, I don't know where your extrinsic and intrinsic files come from. 
They are not produced by the code you sent me.
And I don't understand why you don't simply use Pose2Sim.calibration().

Can you give me a clear run-down of your problem and why you go about it this way?

If it is not related to Pose2Sim, you may be better off asking the question on stackoverflow with the #opencv tag.

harshitsinghcode, 2024-06-05:
And in the parameters what to put, pose2sim.calibration() -- like what to give the parameter to it? inside this how to get that config.toml file for us ...like it's already been given, but I want to use my data from the folder in which I've stored photos of check boards.

davidpagnon, 2024-06-05:
There is nothing to put in the parenthesis, all the parameters come from the Config.toml file. Try the Demo, follow the tutorial afterwards to run it on your own data, and if it still does not work try what I asked you above.

harshitsinghcode, 2024-06-19:
@davidpagnon , I tried with that code for calibration using opencv, didnt give the required result but then i proceeded with your way of -  Pose2Sim.calibration(), and the it shows this - 
[image]
and also my version i checked is - Version: 0.8.3 - 
[image]

davidpagnon, 2024-06-19:
Did you try setting "show_detection_intrinsics" to true in Config.toml as advised in the error message? The checkerboard might not be compliant with the OpenCV requirements:
[image]

harshitsinghcode, 2024-06-20:
> [hmejbri](/hmejbri)

harshitsinghcode, 2024-06-20:
> [hmejbri](/hmejbri),
how did you deal with this error?

harshitsinghcode, 2024-06-20:
im getting this and it's not detecting the corners - 
[image]
@davidpagnon !

davidpagnon, 2024-06-20:
Okay I think your problem is that your margin is not wide enough. You can print a checkerboard here: https://calib.io/pages/camera-calibration-pattern-generator

harshitsinghcode, 2024-06-24:
@davidpagnon , you said it right, i used that checkboard, but now the problem arises with the error- 

with 10 images -  4.973 px
with 24 images - 11.47 px
with 34 images - 12.93 px

harshitsinghcode, 2024-06-24:
[image]

the checkboard i'm using - 

[image]

code - 

[image]

harshitsinghcode, 2024-06-24:
yupp, now it came down to 0.737 px after adjusting it well!

davidpagnon, 2024-06-24:
Nice! So is it working fine now?

harshitsinghcode, 2024-06-24:
but it is not getting under 0.5 px

davidpagnon, 2024-06-24:
Don't worry, it should be fine anyway. 
0.5 px is given just as a scale,  so that people know it is not worth trying to go as low as 0.01 px, and that 5 px is definitely not good enough.

harshitsinghcode, 2024-06-24:
and just for notifying, my extrinsic calib error is getting around -  [17.567] px, which corresponds to [1.376] mm.

harshitsinghcode, 2024-06-24:
Synchronization error - [ ValueError: The length of the input vector x must be greater than padlen, which is 9. ]

[image]

davidpagnon, 2024-06-24:
Wait I think I'm just getting it now: do you only use one single camera?
Pose2Sim currently only works with at least 2 cameras. I am thinking of supporting one single camera but it is a different paradigm so I'm not sure when I'll get to it.

In the meantime, you can try another tool I developed: https://github.com/davidpagnon/Sports2D
It is, of course, less precise, and dedicated to movements performed in a plane (from the side or the front).

harshitsinghcode, 2024-06-25:
Okay, so today i tried using 2 cameras, and still the error exists -

[image]

harshitsinghcode, 2024-06-25:
@davidpagnon

davidpagnon, 2024-06-25:
Did you do pose estimation first? Can you check that the json files are not empty?

Sorry I'm not sure I'll be able to spend much time on it, at least until Friday :/

harshitsinghcode, 2024-06-26:
yupp, the first step i've done is the pose estimation  only through openpose!

harshitsinghcode, 2024-06-26:
just letting you know that i've done the extrinsic calib through checkboard and not through scene.  maybe could that lead to an error to this?

davidpagnon, 2024-06-27:
Calibration is not needed for synchronization, so it's not the problem there.
You did not tell me if your json files were not empty? How many of them do you have and what is inside? (Open them with any text Editor)

harshitsinghcode, 2024-06-28:
So, i assume you are talking about the json file obtained from the open pose rt? .... yeah, it's not empty. Or any other json file to be looked upon?!

harshitsinghcode, 2024-06-28:
But then again, how come the synchronisation is fetching those json, it's not specified in the script as such!

davidpagnon, 2024-06-28:
Yes that's the json files I was talking about. Synchronization is fetching them from like 319. 
https://github.com/perfanalytics/pose2sim/blob/21a5e7742b427967e7b7caa21960739026ed29f5/Pose2Sim/synchronization.py#L319

harshitsinghcode, 2024-07-01:
ohkay, i guess i got it. so i should insert the json files in the pose folder in the trial folders right? -- but what are the next steps, like if i set up two cameras even and i put the openpose jsons in the single trial also balance trial also >? -- like i didnt get the process. could you please run down !?

harshitsinghcode, 2024-07-01:
@davidpagnon

davidpagnon, 2024-07-02:
Hi, I saw your post, I saw the bump up, and I also saw your WhatsApp message, but they were all within a single day and unfortunately I cannot drop everything ti answer, especially these days with how much work I have otherwise 😅

If you just run it in the single trial folder, you should first do:
- calibration, then 
- pose estimation (it will be natively supported without the need for openpose by the end of the week), then 
- synchronization if your cameras are not already synchronized (can be a little tricky, you will need to go by trial and error and read the doc carefully), then
- person identification (unless one single person is visible by all cameras) , then
- triangulation 
- optionally filtering and marker augmentation 
- and then OpenSim scaling (select a few frames from the tec where the subject is well detected)
- then OpenSim inverse kinematics 

However, all that should be fully explained in the Readme.md file. If there is anything's confusing there, please tell me!

harshitsinghcode, 2024-07-02:
haha, this helped me to get more clarity - in the triangulation, i'm getting the error - 

[image]

harshitsinghcode, 2024-07-02:
@davidpagnon , i used the checkboard only in the extrinsic calib, would it be prob?

harshitsinghcode, 2024-07-02:
triangulation done = 
[image]

but we are not able to run this - 
[image]

harshitsinghcode, 2024-07-02:
@davidpagnon , 2 cameras were used only and its shows this - 
[image]

harshitsinghcode, 2024-07-02:
@davidpagnon , this was personassociation - 
[image]

davidpagnon, 2024-07-02:
There may be an issue with your synchronization; as I said it can be a bit tricky to find the good parameters. Since the error could come from any other stage, could you divide your video in frames, remove some to make sure a specific event (foot step for example) happens at the same frame, and then reassemble them into a synchronized video? Ffmpeg is a good tool for that, and stack overflow will help you find the right commands.

I'm just thinking of it now, are you sure all cameras have the same frame rate?

harshitsinghcode, 2024-07-10:
Yupp, the issues got resolved. We used 3 cameras ( 1 android + 1 iphone + 1 dslr ), and then finally the triangulation happened correctly! Will reach out to you for further developments if we think of any in this field.

Thanks for your valuable time David!

closed 2024-07-10 by harshitsinghcode

davidpagnon, 2024-07-10:
Awesome!

This gives me the occasion to tell you that the latest Pose2Sim version was released just a few seconds ago, with the integration of pose estimation. 
Do `pip install pose2sim --upgrade` to access it :)