perfanalytics/pose2sim #109: Calibration keeps restarting

hmejbri, 2024-06-04:
Im trying to perform camera calibration and the intrinsics keeps restarting even though the intrinsics error is less than 0.5px, this is the output : 

```
Calculating intrinsic parameters...
INFO:root:
Camera int_cam_2_img:
INFO:root:cam2_36_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_72_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_108_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_144_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_180_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_216_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_252_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_288_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_324_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_360_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:Intrinsics error: 0.485 px for each cameras.
INFO:root:
Camera int_cam_1_img:
INFO:root:cam1_36_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_72_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_108_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_144_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_180_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_216_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_252_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_288_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_324_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam1_360_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:Intrinsics error: 0.482 px for each cameras.
INFO:root:
Calculating extrinsic parameters...
INFO:root:
Camera int_cam_2_img:
INFO:root:cam2_36_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_72_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_108_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_144_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_180_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_216_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_252_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_288_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_324_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam2_360_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:Intrinsics error: 0.417 px for each cameras.
INFO:root:
Camera int_cam_1_img:
INFO:root:cam1_36_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
```

davidpagnon, 2024-06-04:
This is strange. Does it work with the Demo data? (make sure to set calibration_type = 'convert' and overwrite_intrinsics = true).

hmejbri, 2024-06-04:
> This is strange. Does it work with the Demo data? (make sure to set calibration_type = 'convert' and overwrite_intrinsics = true).

You mean calculate ?

davidpagnon, 2024-06-04:
Sorry yes!

hmejbri, 2024-06-04:
Yes calibration_type = 'calculate' and overwrite_intrinsics = true, what i'm suspecting is 0.48 intrinsics error is getting rounded up to 0.5 which results in restarting calibration

davidpagnon, 2024-06-04:
I don't remember having set up a loop like that, but it is possible that I have introduced a bug in the last version. Are you using the last Pose2Sim version? (Do `pip show pose2sim`)

My question is, does it also loop with the Demo files?

hmejbri, 2024-06-04:
I tried with the demo files and after finishing the intrinsics calculations it says 'Calculating extrinsic parameters' but still displays the instrinsics frames
```
Calculating intrinsic parameters...
INFO:root:
Camera int_cam_3_img:
INFO:root:cam03_01_int.jpg: Corners found.
INFO:root:cam03_02_int.jpg: Corners found.
INFO:root:cam03_03_int.jpg: Corners found.
INFO:root:cam03_04_int.jpg: Corners found.
INFO:root:cam03_05_int.jpg: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam03_06_int.jpg: Corners found.
INFO:root:cam03_07_int.jpg: Corners found.
INFO:root:Corners were detected only on 6 images for camera int_cam_3_img. Calibration of intrinsic parameters may not be accurate with fewer than 10 good images of the board.
INFO:root:Intrinsics error: 0.271 px for each cameras.
INFO:root:
Camera int_cam_2_img:
INFO:root:cam02_01_int.jpg: Corners found.
INFO:root:cam02_02_int.jpg: Corners found.
INFO:root:cam02_03_int.jpg: Corners found.
INFO:root:cam02_04_int.jpg: Corners found.
INFO:root:cam02_05_int.jpg: Corners found.
INFO:root:cam02_06_int.jpg: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam02_07_int.jpg: Corners found.
INFO:root:Corners were detected only on 6 images for camera int_cam_2_img. Calibration of intrinsic parameters may not be accurate with fewer than 10 good images of the board.
INFO:root:Intrinsics error: 0.255 px for each cameras.
INFO:root:
Camera int_cam_4_img:
INFO:root:cam04_01_int.jpg: Corners found.
INFO:root:cam04_02_int.jpg: Corners found.
INFO:root:cam04_03_int.jpg: Corners found.
INFO:root:cam04_04_int.jpg: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam04_05_int.jpg: Corners found.
INFO:root:cam04_06_int.jpg: Corners found.
INFO:root:cam04_07_int.jpg: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:Corners were detected only on 5 images for camera int_cam_4_img. Calibration of intrinsic parameters may not be accurate with fewer than 10 good images of the board.
INFO:root:Intrinsics error: 0.302 px for each cameras.
INFO:root:
Camera int_cam_1_img:
INFO:root:cam01_01_int.jpg: Corners found.
INFO:root:cam01_02_int.jpg: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam01_03_int.jpg: Corners found.
INFO:root:cam01_04_int.jpg: Corners found.
INFO:root:cam01_05_int.jpg: Corners found.
INFO:root:cam01_06_int.jpg: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:cam01_07_int.jpg: Corners found.
INFO:root:Corners were detected only on 5 images for camera int_cam_1_img. Calibration of intrinsic parameters may not be accurate with fewer than 10 good images of the board.
INFO:root:Intrinsics error: 0.266 px for each cameras.
INFO:root:
Calculating extrinsic parameters...
INFO:root:
Camera int_cam_3_img:
INFO:root:cam03_01_int.jpg: Corners found.
```

davidpagnon, 2024-06-05:
Hi, I tried it again and I don't see any bug.

**Can you do `pip show pose2sim` and check that we both work on version 0.8.3?**

I should probably add a carriage return, but you can see that in the first loop it is calculating intrinsic parameters, while in the second one it works on extrinsic ones. (I also need to explicitly sort the folders by alphabetical order, but that is not the issue).

Now, what I cannot figure out is why it is still working on intrinsic parameters while in the extrinsic loop. I don't have this problem on my side.

**Did you edit the code at some point?**
**In [calibration.calculate.extrinsics], did you set 'board' or 'scene'?**

hmejbri, 2024-06-05:
I upgraded pose2sim and the last problem got fixed and a new problem risen. I used demo data but for extrinsics calculations i used the same frames of intrinsics and i set [calibration.calculate.extrinsics] to 'board' (one frame for each camera), the problem is it only displays the cam3 frame then it stops.

hmejbri, 2024-06-05:
I upgraded pose2sim and the previous problem disappeared but a new one appeared.

Im using the demo data and im trying to calculate extrinsics parameters using the 'board' instead of 'scene' so i used one frame from the intrinsics for each camera. The problem is that the extrinsics calculations keeps stopping after only one frame (After the intrinsics gets calculated).

davidpagnon, 2024-06-06:
Hi @hmejbri, 
I tried the Demo calibration again and did not face any issue. Here is what I changed:
- In Config.toml:
  calibration_type = 'calculate'
  extrinsics_method = 'board'
  extrinsics_extension = 'jpg'
- I copied the S00_Demo_BatchSession\S00_Calibration\intrinsics folder into the intrinsics one

Did you do anything differently?

davidpagnon, 2024-06-06:
I think that's what as well. Did you make sure the "extrinsic_extension" was right?
Another hypothesis, did you close the window instead of hitting 'Y'?

davidpagnon, 2024-06-06:
Can you tell me again what's going on, copy-paste the precise error, and if you did anything non standard?

hmejbri, 2024-06-13:
Im still getting the same problem, the extrinsics calculation stops after one camera. I checked the files and everything is fine and with the same extension 'png'.
This is the config file :
```
[calibration]
    calibration_type = 'calculate' # 'convert' or 'calculate'

    [calibration.convert]
    convert_from = 'qualisys' # 'qualisys', 'optitrack', vicon', 'opencap', 'easymocap', 'biocv', 'anipose', or 'freemocap'
        [calibration.convert.qualisys]
        binning_factor = 1 # Usually 1, except when filming in 540p where it usually is 2
        [calibration.convert.optitrack]  # See readme for instructions
        [calibration.convert.vicon]      # No parameter needed
        [calibration.convert.opencap]    # No parameter needed
        [calibration.convert.easymocap]  # No parameter needed
        [calibration.convert.biocv]      # No parameter needed
        [calibration.convert.anipose]    # No parameter needed
        [calibration.convert.freemocap]  # No parameter needed
    

    [calibration.calculate] 
        [calibration.calculate.intrinsics]
        overwrite_intrinsics = true # set to false if you don't want to recalculate intrinsic parameters
        show_detection_intrinsics = true # true or false (lowercase)
        intrinsics_extension = 'png' # any video or image extension
        extract_every_N_sec = 1 # if video, extract frames every N seconds (can be <1 )
        intrinsics_corners_nb = [4,7] 
        intrinsics_square_size = 33 # mm

        [calibration.calculate.extrinsics]
        calculate_extrinsics = true # true or false (lowercase) 
        extrinsics_method = 'board' # 'board', 'scene', 'keypoints'
        moving_cameras = false # Not implemented yet

            [calibration.calculate.extrinsics.board]
            show_reprojection_error = true # true or false (lowercase)
            extrinsics_extension = 'png' # any video or image extension
            extrinsics_corners_nb = [4,7] # [H,W] rather than [w,h]
            extrinsics_square_size = 33 # mm # [h,w] if square is actually a rectangle
```

and this is the log :
```
INFO:root:cam3_104_int.png: Corners found.
INFO:root:cam3_156_int.png: Corners found.
INFO:root:cam3_208_int.png: Corners found.
INFO:root:cam3_260_int.png: Corners found.
INFO:root:cam3_312_int.png: Corners found.
INFO:root:cam3_364_int.png: Corners found.
INFO:root:cam3_416_int.png: Corners found.
INFO:root:cam3_468_int.png: Corners found.
INFO:root:cam3_520_int.png: Corners found.
INFO:root:cam3_572_int.png: Corners found.
INFO:root:Intrinsics error: 0.181 px for each cameras.
INFO:root:
Camera int_cam_2_img:
INFO:root:cam2_42_int.png: Corners found.
INFO:root:cam2_84_int.png: Corners found.
INFO:root:cam2_126_int.png: Corners found.
INFO:root:cam2_168_int.png: Corners found.
INFO:root:cam2_210_int.png: Corners found.
INFO:root:cam2_252_int.png: Corners found.
INFO:root:cam2_294_int.png: Corners found.
INFO:root:cam2_336_int.png: Corners found.
INFO:root:cam2_378_int.png: Corners found.
INFO:root:cam2_420_int.png: Corners found.
INFO:root:cam2_462_int.png: Corners not found. To label them by hand, set "show_detection_intrinsics" to true in the Config.toml file.
INFO:root:Intrinsics error: 0.192 px for each cameras.
INFO:root:
Camera int_cam_1_img:
INFO:root:cam1_36_int.png: Corners found.
INFO:root:cam1_72_int.png: Corners found.
INFO:root:cam1_108_int.png: Corners found.
INFO:root:cam1_144_int.png: Corners found.
INFO:root:cam1_180_int.png: Corners found.
INFO:root:cam1_216_int.png: Corners found.
INFO:root:cam1_252_int.png: Corners found.
INFO:root:cam1_288_int.png: Corners found.
INFO:root:cam1_324_int.png: Corners found.
INFO:root:cam1_360_int.png: Corners found.
INFO:root:cam1_396_int.png: Corners found.
INFO:root:Intrinsics error: 0.313 px for each cameras.
INFO:root:
Calculating extrinsic parameters...
INFO:root:
Camera ext_cam_3_img:
INFO:root:cam3_ext.png: Corners found.
```

davidpagnon, 2024-06-13:
Hi @hmejbri, 
Sis you post and then deleted your comment?
I don't see what the problem is, could you maybe send me your data so that I can try it out?

hmejbri, 2024-06-13:
Yes I deleted the comment because the problem got fixed.

closed 2024-06-13 by hmejbri