perfanalytics/pose2sim #112: Structure of a "Session"

ANaaim, 2024-06-28:
Some thought about what should be a Session after the discussion we had in the seminar : 

   Session_mesure (Generally a subject)
       │
       ├──calibration
       │    ├── Config.toml
       │    ├── Calib_qualisys.toml (if calibration with qualysis)
       │    └── Calib_board.toml (if calibration with video)
       │
       │
       │        
       ├── Acqusition_001
       │    ├── heatmaps
       │    ├── pose
       │    ├── pose-3d
       │    └── videos
       │
       ├── Acqusition_002
       │    ├── heatmaps
       │    ├── pose
       │    ├── pose-3d
       │    └── videos
       |
      ├──..... 
       |
      ├── Acqusition_XXX
      │    ├── heatmaps
      │    ├── pose
      │    ├── pose-3d
      │    └── videos
      │
      ├──Config.toml
      └──User (a folder for some logging)

Currently or we have everything in each acqusition or we have to have one folder supplementary above. 
To have such organisation it might be needed in some function to change the detection of the session
from  :
````python
session_dir = os.path.realpath(os.path.join(project_dir, '..','..'))
````
to :
```python 
session_dir = os.path.realpath(os.path.join(project_dir, '..'))
 ````

Would you agree ?

ANaaim, 2024-06-28:
Are you ok with it ? If yes I can do do a PR.  :)

davidpagnon, 2024-06-28:
Hi, thank you for the proposal!

Sorry, I could not find time to think about it before :/ My mind is not very sharp at that time of the night but it looks good, and I would be happy if you did a PR!

From what I can think of the top of my head, the last option looks like It's the most convenient in terms of coding and of functionalities.

Do we need a User folder though? The logs could stay in the session directory, couldn't they? Also, I'm not sure we need a Config.toml file in the calibration folder.

It would be nice if there was a Config.toml file in each Acquisition folder as well, that would be taken into account only if the parameters are uncommented. 
- This way we could run some Acquisition with different parameters of needed, and still batch process all of them. 
- And it would also be cool to be able to run Pose2Sim in an acquisition folder, so that we can run it for one single trial if needed 

We can always call each other to make sure we are on the same page.

davidpagnon, 2024-06-28:
Also, note that @hunminkim98 and I are about to implement pose estimation. I'm running the last tests but it should be released sometime next week. We may run into some merge issues but it should not be too hard to solve.

ANaaim, 2024-06-29:
From what you say, it also could be in option as an int. it would indicate what is the parent level.
0 : the acquisition itself 
1 : the participant level
2 : the session level

?

davidpagnon, 2024-07-01:
Do you mean adding the option in the Config file?
If it could be determined automatically it would be even better :)
(but I'm not sure I understood)

davidpagnon, 2024-07-01:
Actually I think I understood, and I agree!

ANaaim, 2024-07-01:
And if not given ==> the "legacy" behavior ?

davidpagnon, 2024-07-01:
What I had in mind is that it would be cool if it could guess it automatically from the folder structure.

But honestly, it would be nice but I'm not sure we actually need the legacy option. The only thing is that it should be possible to run it from within the acquisition (single analysis) or from the session (batch analysis). And it would be nice to have a Config file in each acquisition session so that the user can uncomment and edit a parameter if they want to analyze this acquisition slightly differently.

Does it make sense?

davidpagnon, 2024-07-01:
(I don't have your number but mine can be found on the Internet, so don't hesitate to give me a call :) )

davidpagnon, 2024-07-09:
Hi, sorry for being so long; I made significant changes that would have been a pain to merge. 
But if you are still up for doing the folder architecture, it should be alright now! Now that I think about it, it is not worth keeping the complicated one for legacy, I don't think anyone actually found it convenient. 

Note that Pose estimation is now available, but only if you build from source. I need to wait for the author of RTMlib to release the changes we made to his library in order to obtain feet keypoints along with body (and without all the face and finger ones):
```
git clone --depth 1 https://github.com/perfanalytics/pose2sim.git
cd pose2sim
pip install .
```

davidpagnon, 2024-07-10:
(update: no need to install from source anymore, the author of RTMlib just pushed his last release on Pypi, which lets me push the last Pose2Sim release as well)

ANaaim, 2024-07-11:
@hunminkim98 @davidpagnon    thanks :)

davidpagnon, 2024-07-11:
Okay, thanks! I just tested it for the "standard" structure:
```
Single session
├── calibration /
├── videos /
└── Config.toml
```

Do your modifications work for the batch structure without further change in Pose2Sim.py -> read_config_files()? (and is it equivalent to this one?)
```
Batch session
├──calibration /
├── acquisition_001 /
│    ├── videos /
│    └── Config.toml
├── acquisition_002
│    ├── videos /
│    └── Config.toml
└──Config.toml

```

ANaaim, 2024-07-12:
You need to have a config file in each acquisiton to do the batch processing ? If yes that might come from this, but I doubt as in  the code that i am modifying i am saying that it should only look 1 folder above instead of two 

Thanks in advance.

davidpagnon, 2024-07-17:
Hi, I just fixed batch processing, created a demo for it, and updated the documentation :)

ANaaim, 2024-07-18:
Will try it on my example tomorrow !!

davidpagnon, 2024-07-19:
Be aware that for now you need a Config.toml file in both root and trial levels, so that you can
- set specific parameters for specific trials
- run either from the trial folder to run a single trial, or from the root folder to batch process 

 The Config.toml files of the trials can all be empty (or fully commented) if they can all be processed the same way. There is probable a smarter way to do it that would not require a Config file everywhere, but I really need to work on other things for a little while 😅

closed 2024-09-03 by davidpagnon