perfanalytics/pose2sim #115: Issues during extrinsic calib

peterlololsss, 2024-07-08:
Hello there,

I have the checkerboard images stored in the each camera folder under extrinsic folder, and after running Pose2sim.calibration() 
command I successfully got Points deteciton result, but after I submit the first one, 
the error pops up :
    operands could not be broadcast together with shapes (8,1,2) (0,)

in the config file i set like this:
 # Camera placements, need to be done before every session
      [calibration.calculate.extrinsics]
      calculate_extrinsics = true # true or false (lowercase) 
      extrinsics_method = 'board' # 'board', 'scene', 'keypoints'
      # 'board' should be large enough to be detected when laid on the floor. Not recommended.
      # 'scene' involves manually clicking any point of know coordinates on scene. Usually more accurate if points are spread out.
      # 'keypoints' uses automatic pose estimation of a person freely walking and waving arms in the scene. Slighlty less accurate, requires synchronized cameras.
      moving_cameras = false # Not implemented yet

         [calibration.calculate.extrinsics.board]
         show_reprojection_error = true # true or false (lowercase)
         extrinsics_extension = 'png' # any video or image extension
         extrinsics_corners_nb = [9,6] # [H,W] rather than [w,h]
         extrinsics_square_size = 80 # mm # [h,w] if square is actually a rectangle

The intrinsic calibration is fine but i am stuck at extrinsic calibraiton, I also tried to change cameras but the error message results the same after submitting the first image(typed Y)

I also tried to switch to scene method, but still after choosing points, the error message is about operands could not be broadcast together :(

I'm rly at loss here

closed 2024-07-09 by peterlololsss

davidpagnon, 2024-07-09:
Hi @peterlololsss, I see that you marked this issue as completed. Can you tell me how you fixed it, for future reference?

peterlololsss, 2024-07-10:
> Hi @peterlololsss, I see that you marked this issue as completed. Can you tell me how you fixed it, for future reference?

Well for some reason the calibration.py I installed is out of date(the annotation on that file was version 0.4), so once I updated that file the issue is gone.

I am still not sure how I pip installed a much older version of the project, but just to be sure I reconfigured my environment and the issue I had never occurred again

davidpagnon, 2024-07-10:
Okay, I'm glad it's working again then!