perfanalytics/pose2sim #117: Did the expected format of json change with latest 0.9 version ?

ANaaim, 2024-07-11:
Hi, 

I was wondering if the format of the expected json was changed in latest version (0.9)? 

Best regards,

davidpagnon, 2024-07-11:
It shouldn't do you experience any issue?

The main difference is that we don't use body_25 or body_25B openpose models anymore, but Halpe_26, which uses a different keypoint ordering.

davidpagnon, 2024-07-11:
You're right, I messed it up, thank you for spotting it!
I'll fix it right now

ANaaim, 2024-07-11:
Cool. so no need to copy the data from a the pose folder to a pose-associated ?

davidpagnon, 2024-07-11:
Can you confirm that it works with the last patch on triangulation.py?
You should not have to copy anything, no.

ANaaim, 2024-07-11:
i should do a pip update pose2sim ?

ANaaim, 2024-07-11:
I will test this.

ANaaim, 2024-07-11:
it work with the new version of triangulation. Let me know when it is integrated in the pip install

closed 2024-07-11 by davidpagnon

ANaaim, 2024-07-12:
Hi David, did you integrate the modif in a new version that can be installed with pip ? 

Thanks.

davidpagnon, 2024-07-13:
Hi, 
Sorry I initially wanted to wait until more changes were made but it does not make sense, I just pushed the last release to pypi