perfanalytics/pose2sim #118: Does calibration with a board put the Y in the wrong direction

ANaaim, 2024-07-11:
Other question  :

When using board calibration, the data are ... upside down. Is it the expected behavior ? 

[image]

Sorry for all the question, I am finishing a V1 for my tool-box ^^ so... a lot of use case

davidpagnon, 2024-07-11:
Definitely not expected, but it can happen depending on the pattern of the checkerboard. I have not been able to check, but if I believe what I found on the internet, the X axis goes from the black corner to the other black corner, and the Y axis from the black corner to the white one. So if you have something like in the picture below, you end up with a Z axis going underground.

[image]

The solution to this would be to print another checkerboard, mirrored from this one.
I'm not a 100 percent sure of this so I'd be grateful if you can confirm!

(Note that in any case, you need to have a number of rows different from the number of lines, otherwise it will not be rotationally invariant and axes will be attributed randomly).

ANaaim, 2024-07-11:
Ok 👍 I Will investigate :)

closed 2025-05-15 by davidpagnon