perfanalytics/pose2sim #119: I have a question about the Marker Augmentation input data.

omnikhk, 2024-07-12:
It is indicated that at least these locations were required for marker augmentation.

["Neck", "RShoulder", "LShoulder", "RHip", "LHip", "RKnee", "LKnee",
 "RAnkle", "LAnkle", "RHeel", "LHeel", "RSmallToe", "LSmallToe",
 "RBigToe", "LBigToe", "RElbow", "LElbow", "RWrist", "LWrist"].

Is this the center point of each location where 2D keypoints are triangulated into 3D?
Or, Where the marker was tracked?

davidpagnon, 2024-07-12:
Hi, I'm not sure I understand the question. The LSTM needs to have certain inputs, otherwise it fails. Kind of like you cannot solve x = a*b if you don't give a and b as inputs.

And these inputs are the triangulated keypoints with the names you provided. Some 2D pose models, such as Coco 17 for example, don't have feet keypoints, so it would not work.

However, note that we sometimes have better results without marker augmentation, so it may be worth considering not using it.

davidpagnon, 2024-07-29:
Closed as stale. Feel free to reopen the issue if you do not consider it has been solved.

closed 2024-07-29 by davidpagnon