perfanalytics/pose2sim #121: Model is rotated by 90° in Opensim

PockyStiks, 2024-07-16:
Hello,
When visualizing my captured motion in Opensim, the motion plays out correctly, but the model is rotated by 90° facing the floor, and the pelvis_tilt and pelvis_list values are all over the place. I'm using 2 720p webcams with RTMPose alongside the provided LSTM model, scaling and IK files. The only notable changes made to the Config.toml file are participant height, video file formats (my recordings are .avi), and checkerboard dimensions (I'm using a 4x5 35mm board for both intrinsics and extrinsics, I also tried a 4x7 22mm board but got the same result). The RMS calibration errors for each camera are 1.373px (2.532mm) and 1.497px (2.355mm). I should also mention that when running the single person balancing trial demo (with calibration_type set to calculate) the simulation ran correctly. Any idea on what could be causing the model to rotate? Any suggestions would be greatly appreciated :) 

https://github.com/user-attachments/assets/25b666d7-e430-4762-8484-e378f65691b6

davidpagnon, 2024-07-16:
Hi, I have the feeling your checkerboard was set vertically like in OpenCap, instead of horizontally.

If so, you would just need to swap Y and Z coordinates in your object points in config.toml. Unfortunately, I have not thought of making that use possible for the "extrinsics with a checkerboard" option, but I can probably implement it in the near future if you think it could be useful.

In the meantime, maybe you can convert your TRC file with this script?
https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/Utilities/trc_Zup_to_Yup.py

PockyStiks, 2024-07-17:
Thank you for the quick response! Indeed the issue was caused by me holding the checkerboard vertically instead of putting it flat on the floor. After running the script, the model was positioned correctly but the skeleton got a little deformed. With my current setup it is no doubt easier to calibrate using Y as up, so adding that option would definitely be useful, but isn't in any way urgent. Thanks again for your help :)