perfanalytics/pose2sim #122: Anaconda Prompt outputs: "QGuiApplication::font(): no QGuiApplication instance and no application font set."

djinayon, 2024-07-21:
Hi, I'm very new to pose2sim and am currently trying out the demonstration part 1. The prompt (Anaconda) outputs 'no QGuiApplication instance and no application font sent' whenever I get to the filtering line and then exits ipython right after (see image below). I tried running the codes line by line and it still shows the same result. I can't get to the marker augmentation and opensim processing part due to this. Is this normal? If not, what might be the problem here? I tried reading forums but I can't seem to find anything that resolves this problem.

[image]

davidpagnon, 2024-07-21:
Hi, unfortunately it is a know issue and I have to found a way to solve it yet: https://github.com/perfanalytics/pose2sim/blob/ac037e05cef4f62a42c0662c2390796cf05d046b/Pose2Sim/Demo_SinglePerson/Config.toml#L156
Note that it has also been added to the [to-do list](https://github.com/perfanalytics/pose2sim/issues/40#issuecomment-1783120976). If anyone comes across it and feels like tackling the issue, feel absolutely free and let me know!

In the meantime, a simple solution would simply be to set `display_figures = false` in your Config.toml file.

djinayon, 2024-07-21:
Hi, thank you for your fast response. I tried the solution and  then proceeded to part 2. However, I encountered this problem (see image below). I've read the thread relating to this but could not quite understand the solution.  Can you enlighten me?

[image]

davidpagnon, 2024-07-21:
Oh yes, you're right, I changed the folder structure and apparemment did not edit the paths in the setup files. Just change the path of your trc file to the one up just generated (same for scaling and inverse kinematics).

Note that in the near future, this will be done in the command line by default, with Pose2Sim.inverseKinematics(). (You will still be able to use OpenSim GUI if your results are tricky or you need to work by trial and error with a visual feedback).

djinayon, 2024-07-21:
> Oh yes, you're right, I changed the folder structure and apparemment did not edit the paths in the setup files. Just change the path of your trc file to the one up just generated (same for scaling [...]

Thanks, I did not think of that. I just manually added the markers as pointed out in #69. I can't seem to find the files for the inverse kinematics though. Nothing comes up in the Pose2Sim folder (see 2nd image). I'm assuming they should either be included in the cloning or be generated from the previous part so I should be able to find it within the folder right?

[image]

[image]

davidpagnon, 2024-07-21:
Sorry, same problem of setup file that was not edited'. For now you need to either:
- set the path to your .mot "motion file" in the IK tool
- or run IK, then go to the left pane of the main window, right click on "Results", and save it

djinayon, 2024-07-21:
> Sorry, same problem of setup file that was not edited'. For now you need to either:
> 
> * set the path to your .mot "motion file" in the IK tool
> * or run IK, then go to the left pane of the main window, right click on "Results", and save it
The problem seems to be that I don't have the .mot file (see image below). Is the .mot file generated from the first part?

[image]

davidpagnon, 2024-07-21:
The mot file is the result of Inverse kinematics: either you specify the output path directly in the IK tool (in your screenshot, the field is blank, you need to add a path); or you don't, and then on the left pane, under your scaled model, you right click results and "save as" a mot file.

I kind of repeated myself but I'm not sure how to say it otherwise, or if I just did not understand you :)

djinayon, 2024-07-21:
> The mot file is the result of Inverse kinematics: either you specify the output path directly in the IK tool (in your screenshot, the field is blank, you need to add a path); or you don't, and then [...]

I probably just need some more exploring before I fully understand what you meant (because I'm a total newbie haha). I'll explore for now so as not to take much of your time. Thanks again for responding :)

davidpagnon, 2024-07-21:
If you ran inverse kinematics and you can see the skeleton move, you definitely have the answer above. If not, that may be another issue but I would be surprised (but life is full of surprises).

djinayon, 2024-07-21:
> If you ran inverse kinematics and you can see the skeleton move, you definitely have the answer above. If not, that may be another issue but I would be surprised (but life is full of surprises).

I just understood how to run the inverse kinematics haha (yea, it's me being a total newb). Thanks again, I'll pop right in if I encounter other issues. Your active responses are very much appreciated :)

closed 2024-07-29 by davidpagnon