perfanalytics/pose2sim #123: Question about marker location

CrazyEpi, 2024-07-29:
Hi. Thank you for your great work! My team is trying to use Pose2Sim to gather muscle data for our project. 

We have ran through the overall process with a 4-camera system. However, our goal is to generate arm muscle data and the .trc file for LSTM model obviously cannot be used for other models. 

Thus, I want to ask that if there is a way to change the marker locations in pose2sim, so we can use it for other models. 

We already tried to modify .osim file, but honestly it's quite challenging when it comes to third party models.

Thank you for reading this issue and looking forward to your reply.

davidpagnon, 2024-07-29:
Hi, there are several steps to undergo, and it depends what's your goal:
- first, you need to use a different model for pose estimation. You can train your own model with [DeepLabCut](https://www.mackenziemathislab.org/deeplabcut) or [MMPose](https://mmpose.readthedocs.io/en/latest/demos.html), or take one that is already supported by RTMPose (see [model zoo here](https://github.com/open-mmlab/mmpose/tree/main/projects/rtmpose#-model-zoo-)). You will need to convert the output to the OpenPose json format (you can use this script if you run [DeepLabCut](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/Utilities/DLC_to_OpenPose.py))
- then you need to give each keypoint ID a name so that the markers have the name you expect. You can either hardcode it in [skeletons.py](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/skeletons.py), or write it in the [Config.toml](https://github.com/perfanalytics/pose2sim/blob/3bb2c5dfb6a94fc2d19d24b0cb87a4395ce92672/Pose2Sim/Demo_SinglePerson/Config.toml#L204) file.
- And then, you can use your own OpenSim model. I would manually place the points on the unscaled skeleton using the GUI. In my case, I captured 3D meshes of the people + marker data, which helped me optimize the locations of the triangulated keypoints with regards to true joint centers on the model. But you will cross that bridge when you come to it :)

Am I answering your question?

CrazyEpi, 2024-07-31:
Thank you for your info!

If it's ok, I'll try to join you guys in late august, I guess I'll be freer then.

closed 2024-07-31 by CrazyEpi