perfanalytics/pose2sim #126: Guidance on how to use pose2sim

keyo030, 2024-08-08:
I hope this message finds you well.

I am currently working with OpenSim and during my research I came across your GitHub repository for pose2sim and was truly impressed by your work. As I am relatively new to both OpenSim and Python, I would greatly appreciate your assistance with a few questions regarding your tool.

Specifically, I am trying to use a file with OpenSim to stimulate my model. From what I understand, we need to convert a .csv file to a .trc file to ensure compatibility and enable inverse kinematics. However, I am unsure of the following:
1. Which function should I use to convert the .csv file to a .trc file (my data has been attached to this post)
[image] ?
2. Are there any additional files or steps required to perform inverse kinematics in OpenSim?

If you could provide some guidance or a walkthrough on how to achieve this conversion, I would be extremely grateful.

Thank you very much for your time and consideration.

davidpagnon, 2024-08-08:
Hi, thanks for the appreciation!

How did you obtain this CSV file?
Pose2Sim gives you a trc file that can be used directly withing OpenSim. This TRC file has a simple header, and then the 3D coordinates of body markers.

If you transform your CSV file so that it is consistent with a trc file, with keypoint names that correspond to the markerset in your OpenSim model, this should work.

I would advise to follow the tutorial in the Readme file: https://github.com/perfanalytics/pose2sim
You can also try the first 3 tutorials of OpenSim: https://opensimconfluence.atlassian.net/wiki/spaces/OpenSim/pages/53088700/Tutorial+1+-+Intro+to+Musculoskeletal+Modeling

keyo030, 2024-08-09:
Hello,

Thank you very much for your reply and the useful links!

I obtained this CSV file from https://archive.ics.uci.edu/dataset/455/simulated+falls+and+daily+living+activities+data+set (an online database).

Which function should I use to convert my .csv file to the .trc format? Also, do I need to use another .XML file to perform inverse kinematics?

davidpagnon, 2024-08-09:
Hi, there is no function to convert it but it should be pretty easy, so that could be your first python project :) Just run the Pose2Sim demo to obtain a trc file and compare them to know what it should look like. (You can also edit it by hand within VScode, Notepad++, or any other text editor).

If the points in your CSV file correspond to the ones in one of the Pose2Sim.osim models, you will just have to change their names in the new trc file. If not, you will also have to place the points in the right position in your osim model.

davidpagnon, 2024-09-03:
This issue has been marked as stale. Please feel free to reopen it if needed!

closed 2024-09-03 by davidpagnon