perfanalytics/pose2sim #127: I have meet some problems while attempting to using my own vdieo to preview the results

ESJIAN, 2024-08-13:
`Exception                                 Traceback (most recent call last)
Cell In[6], line 2
      1 from Pose2Sim import Pose2Sim
----> 2 Pose2Sim.triangulation()

File D:\Conda_dir\pose2sim\lib\site-packages\Pose2Sim\Pose2Sim.py:375, in triangulation(config)
    372 logging.info(f"Project directory: {project_dir}")
    373 logging.info("---------------------------------------------------------------------\n")
--> 375 triangulate_all(config_dict)
    377 end = time.time()
    378 elapsed = end-start

File D:\Conda_dir\pose2sim\lib\site-packages\Pose2Sim\triangulation.py:937, in triangulate_all(config_dict)
    935 # 
    936 if nb_persons_to_detect == 0:
--> 937      raise Exception('No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.')
    939 # IDs of excluded cameras
    940 # id_excluded_cams_tot = [np.concatenate([id_excluded_cams_tot[f][k] for f in range(frames_nb)]) for k in range(keypoints_nb)]
    941 id_excluded_cams_tot = [np.hstack(np.hstack(np.array(id_excluded_cams_tot[n]))) for n in range(nb_persons_to_detect)]

Exception: No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.
`

I cant analysis how do this happend ，I thought it may reason about the videos，but i have try the sample videos and still get the same results..   @...@

ESJIAN, 2024-08-13:
It happend when i enter  
 
`
from Pose2Sim import Pose2Sim
Pose2Sim.triangulation()
`

ESJIAN, 2024-08-13:
[logs.txt](https://github.com/user-attachments/files/16598994/logs.txt)
[Calib_scene.txt](https://github.com/user-attachments/files/16599003/Calib_scene.txt)

here are relative files

hunminkim98, 2024-08-13:
Hello, sometimes I also faced this issue due to following
In my case, it occured by
1. Not match between calibration result and pose data.
2. Calibration result is too bad.
3. Pose data(json) is incorrect.

And I saw your logs.txt and I think this issue from calibration result which is too high reprojection error😢
Also, correlation for camera sync was 0 it is not normal. 

Could you confirm that the data you used for calibration and the pose data (JSON) file are correct?

ESJIAN, 2024-08-14:
[image]

As what you have said,thanks！I  konw how to solve right now 
> Hello, I came across your post and I'm replying because this is an issue I've experienced as well.你好，我看到了你的帖子，我回复你是因为这也是我经历过的问题。 Sometimes, I faced this issue due to following有时，由于以下原因，我面临这个问题
> 
> 1. Not match between calibration result and pose data.标定结果与位姿数据不匹配。
> 2. Calibration result is too bad.校准结果太糟糕了。
> 3. Pose data(json) is incorrect.姿势数据(json)不正确。
> 
> And I saw your logs.txt and I think this issue from calibration result which is too high reprojection error😢我看到了你的log。txt，我认为这个问题来自校准结果，这是太高的重投影误差😢 Also, correlation for camera sync was 0 it is not normal.此外，相机同步的相关性为0，这是不正常的。
> 
> Could you confirm that the data you used for calibration and the pose data (JSON) file are correct?你能确认你校准时使用的数据和姿势数据(JSON)文件是正确的吗?

ESJIAN, 2024-08-14:
[image]
Maybe just as you said,Calibration result is too bad. It might has relationship with Calibration result?
> Hello, I came across your post and I'm replying because this is an issue I've experienced as well. Sometimes, I faced this issue due to following
> 
> 1. Not match between calibration result and pose data.
> 2. Calibration result is too bad.
> 3. Pose data(json) is incorrect.
> 
> And I saw your logs.txt and I think this issue from calibration result which is too high reprojection error😢 Also, correlation for camera sync was 0 it is not normal.
> 
> Could you confirm that the data you used for calibration and the pose data (JSON) file are correct?

ESJIAN, 2024-08-14:
Oh, I understand that when i switch to scene measurements ,I havent change object_coords_3d!!

ESJIAN, 2024-08-14:
I have only two cameras and I guess  there are some parameters I havent adjust

ESJIAN, 2024-08-14:
I switch to use board to calibrate the extrinsic parameters and it looks OK from the preview.But still get the also error report like below
`
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[22], line 2
      1 from Pose2Sim import Pose2Sim
----> 2 Pose2Sim.triangulation()

File D:\Conda_dir\pose2sim_sample\lib\site-packages\Pose2Sim\Pose2Sim.py:374, in triangulation(config)
    371 logging.info(f"Project directory: {project_dir}")
    372 logging.info("---------------------------------------------------------------------\n")
--> 374 triangulate_all(config_dict)
    376 end = time.time()
    377 elapsed = end-start

File D:\Conda_dir\pose2sim_sample\lib\site-packages\Pose2Sim\triangulation.py:937, in triangulate_all(config_dict)
    934 nb_persons_to_detect = len(Q_tot)
    936 if nb_persons_to_detect ==0:
--> 937     raise Exception('No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.')
    939 # IDs of excluded cameras
    940 # id_excluded_cams_tot = [np.concatenate([id_excluded_cams_tot[f][k] for f in range(frames_nb)]) for k in range(keypoints_nb)]
    941 id_excluded_cams_tot = [np.hstack(np.hstack(np.array(id_excluded_cams_tot[n]))) for n in range(nb_persons_to_detect)]

Exception: No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.

`

ESJIAN, 2024-08-14:
My purpose is to test the videos from two same cameras, and I try to search the if any others try this project on CN Internet but none...（T ... T） I will keep trying！

ESJIAN, 2024-08-14:
Oh, I wonder what files I should upload to give you to analysis the question

davidpagnon, 2024-08-14:
It is definitely possible, even if it is not ideal, to use 2 single cameras. Can you send the preview of the extrinsic calibration images?

You may also have an issue with synchronization, it is sometimes good to play with the parameters in the Config file to get good results (or to synchronize by hand)

davidpagnon, 2024-09-03:
This issue has been marked as stale. Please feel free to reopen it if needed!

closed 2024-09-03 by davidpagnon