perfanalytics/pose2sim #129: Automates opensim process

peterlololsss, 2024-08-21:
Hi, thanks for the great work. It helps a lot during my project. 

To save some effort i wrote some matlab functions to automate the process of scaling and ik, and I think I could (at least try to) figure out a similar python function to integrate it into pose2sim pipeline:) (I think opensimProcessing is not implemented right?)

If you're open to it, I would be happy to work on it and submit a pull requests with the changes.

davidpagnon, 2024-08-21:
Hi Peter, 
That would be with pleasure, thanks for offering!
I have a draft for it but did not find time to integrate it, I can send it to you if you want.
Would you mind talking about it on the Discord server? https://discord.com/invite/4mXUdSFjmt

davidpagnon, 2024-09-20:
Thank you @peterlololsss, it is now merged, I edited a few things as explained there: https://github.com/perfanalytics/pose2sim/releases/tag/v0.10.0
And I pushed the release to make it available to everyone!

closed 2024-09-20 by davidpagnon