perfanalytics/pose2sim #132: Can I use other musculoskeletal model instead of Pose2Sim model?

InckieUltra, 2024-09-16:
I noticed the provided model have no muscles. I wonder if I can use other models like "LaiUhlrich2022" which was used in OpenCap, so that I can perform muscle analysis?

davidpagnon, 2024-09-16:
Hi, considering that most body models are based on the Rajagopal 2015 models, you could just put the muscles back from the Lai-Uhlrich model to the Pose2Sim one. Just make sure there is no significant difference between joint definitions.

I initially removed the muscles because it made OpenSim slow at loading models, but I could probably put them back. What do you think?

InckieUltra, 2024-09-16:
Thanks for your kindly reply. 
I think that helps a lot! 👍

maipatrick, 2024-09-18:
Hi,

could you help me introducing valgus varus motion at the knee joint in the current model. My plan it to lock the joint (not allow movement) but still compute joint moments via ID. 
would be great to get some help.

davidpagnon, 2024-09-18:
Hi, this is more of a question you should ask in the OpenSim forum: https://simtk.org/plugins/phpBB/indexPhpbb.php?f=91
There, you'll find people who are way more experts than me :)

In case that can help you phrase the post, the Pose2Sim knee definition is the Rajagopal 2015 one, which is also used in many other opensim models.

davidpagnon, 2024-09-20:
I'm closing the issue, feel free to reopen it if you need it!

closed 2024-09-20 by davidpagnon