perfanalytics/pose2sim #136: Error when running analysis with caliscope sample data

JustHere27, 2024-10-06:
I tried to run an analysis with the [caliscope](https://mprib.github.io/caliscope/) sample data.
I am unclear on what is causing the error.

Any assistance would be greatly appreciated. 

It says that `calib_cam1['matrix'][0][0]` caused `TypeError: 'int' object is not subscriptable`.
However, in calibration config.toml `[cam_1]` had: 
`matrix = [[1086.1181825737008, 0.0, 946.8128791033895], [0.0, 1080.8281918966268, 585.7453083609911], [0.0, 0.0, 1.0]]`

I tried running :
```python
from Pose2Sim import Pose2Sim
Pose2Sim.poseEstimation() # I skipped the calibration as I am using the config.toml fom the sample data
Pose2Sim.synchronization()
Pose2Sim.personAssociation()
Pose2Sim.triangulation()
Pose2Sim.filtering()
Pose2Sim.markerAugmentation()
Pose2Sim.kinematics()
```

It stopt in `Pose2Sim.personAssociation()` with this stacktrace:
```cmd
Single-person analysis selected.
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[1], line 4
      2 Pose2Sim.poseEstimation()
      3 Pose2Sim.synchronization()
----> 4 Pose2Sim.personAssociation()
      5 Pose2Sim.triangulation()
      6 Pose2Sim.filtering()

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:329, in personAssociation(config)
    326 logging.info(f"Project directory: {project_dir}")
    327 logging.info("---------------------------------------------------------------------\n")
--> 329 associate_all(config_dict)
    331 end = time.time()
    332 elapsed = end-start

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\personAssociation.py:752, in associate_all(config_dict)
    748     rewrite_json_files(json_tracked_files_f, json_files_f, proposals, n_cams)
    751 # recap message
--> 752 recap_tracking(config_dict, error_min_tot, cameras_off_tot)

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\personAssociation.py:592, in recap_tracking(config_dict, error, nb_cams_excluded)
    590 calib = toml.load(calib_file)
    591 calib_cam1 = calib[list(calib.keys())[0]]
--> 592 fm = calib_cam1['matrix'][0][0]
    593 Dm = euclidean_distance(calib_cam1['translation'], [0,0,0])
    594 mean_error_mm = np.around(mean_error_px * Dm / fm * 1000, decimals=1)

TypeError: 'int' object is not subscriptable
```

this was my file structure was: 
```
├───calibration
│   └───config.toml
├───videos
│  ├───port_1.mp4
│  ├───port_2.mp4
│  └───port_3.mp4
└───config.toml
```

my calibration config.toml was :
```toml
camera_count = 3
creation_date = 2024-08-18T11:45:12.286017
[... 289 lines cut ...]
     [[pose.CUSTOM.children.children]]
     name = "Head"
     id = 17
        [[pose.CUSTOM.children.children.children]]
        name = "Nose"
        id = 0
     [[pose.CUSTOM.children.children]]
     name = "RShoulder"
     id = 6
        [[pose.CUSTOM.children.children.children]]
        name = "RElbow"
        id = 8
           [[pose.CUSTOM.children.children.children.children]]
           name = "RWrist"
           id = 10
     [[pose.CUSTOM.children.children]]
     name = "LShoulder"
     id = 5
        [[pose.CUSTOM.children.children.children]]
        name = "LElbow"
        id = 7
           [[pose.CUSTOM.children.children.children.children]]
           name = "LWrist"
           id = 9
```

davidpagnon, 2024-10-07:
Hi, 

From what I can see [here](https://utexas-my.sharepoint.com/:x:/r/personal/priblede_my_utexas_edu/_layouts/15/Doc.aspx?sourcedoc=%7BCECE835A-4CB0-45D9-895A-248EF1C86179%7D&file=xy_SIMPLE_HOLISTIC.csv&action=default&mobileredirect=true), Caliscope and Pose2Sim don't provide the same pose estimation output. 
- Pose2Sim output is in the OpenPose json format, and synchronization is done afterwards
- CaliScope output is a custom csv one, with synchronization done beforehand. It is either done automatically with webcams, or requires to create a [frame_time_history.csv](https://utexas-my.sharepoint.com/:x:/r/personal/priblede_my_utexas_edu/_layouts/15/Doc.aspx?sourcedoc=%7BB030D806-CDED-4C54-8B4D-C3ABE6D06D14%7D&file=frame_time_history.csv&action=default&mobileredirect=true) file

Pose2Sim supports Caliscope calibration so you can use it for your calibration if you prefer; but other than that I would encourage you to try both tools independently from the pose estimation stage, and see what best fits your needs.

@mprib may be able to tell you more about it!

davidpagnon, 2024-10-07:
Actually, I just realized that although the calibration correctly loads in the Blender addon: https://github.com/davidpagnon/Pose2Sim_Blender , 
there is a little bug in the recap messages of personAssociation and triangulation. I'm fixing it now and will release a new version by the end of the day, that you will be able to install with `pip install Pose2Sim -U`

NB: Since there is only one person in the scene, you can skip this personAssociation.

davidpagnon, 2024-10-07:
I just released the fix, would you mind telling me if it works on your end?

P.S.: I misread this message and thought you had skipped pose estimation. I was wrong, you did things correctly! (although you can still skip personAssociation() )
```python
Pose2Sim.poseEstimation() # I skipped the calibration as I am using the config.toml fom the sample data
```

davidpagnon, 2024-10-07:
Hi, sorry about that, I tried to bring a quick fix to something that jumped to mind, but without testing it... I'll properly test it tonight and keep you updated!

mprib, 2024-10-07:
Thank you for the mention to bring this to my attention @davidpagnon. I had been unaware of the OpenPose format, just defaulting to my preference for tabular tidy-style data. [Obligatory xkcd](https://xkcd.com/927/).

I've opened an issue to jog my memory to revisit this as time permits: https://github.com/mprib/caliscope/issues/650

davidpagnon, 2024-10-07:
Hi again, 

I had another look at it and upgraded Pose2Sim, it should not throw any error anymore.

Now, here are two more issues:
- poseEstimation() does not work well with images when they are rotated. Until today, I either filmed videos upright, or used cameras that detected their own rotation (phone, action camera...), so I did not have this problem.

  I had a look and there is a button in CaliScope GUI to rotate videos. This is then specified in the calibration file as `rotation_count`. I have not implemented such a feature, but I guess it would be a good idea!

  In the meantime, you can rotate your videos and your calibration file with the script below. 
  
- Results are still far from perfect. Here are my current guesses:
  - I may have made a mistake in the rotation of the calibration file, or in handling the rotation_count parameter in CaliScope calibration
  - Or it could be due to the cameras in the sample being distorted, which I currently do not handle. Undistorting points should be easy to fix, but I have trouble finding time right now. 

  Either way, if you ever have a chance to test Pose2Sim from a Caliscope calibration on other data, I would be quite interested in knowing how it goes!

-----

```python
# ROTATE VIDEOS AND CALIBRATION FILES

# pip install moviepy
# Go to your videos folder and run this script

import os 
from pathlib import Path
import toml
import numpy as np
import cv2
from moviepy.editor import *
from Pose2Sim.common import rotate_cam

videos_files = ['port_1.mp4', 'port_2.mp4', 'port_3.mp4']
rotation = [-90, -90, -90]
calib_file = '../calibration/config.toml'

# Rotate videos
for vid, rot in zip(videos_files, rotation):
    Path.rename(Path(vid), f'{vid}.old')
    clip = VideoFileClip(f'{vid}.old')
    clip = clip.rotate(rot)
    clip.write_videofile(vid)
    clip.close()
    
# Rotate calibration files
calib = toml.load(calib_file)
cal_keys = [c for c in calib.keys() 
            if c not in ['metadata', 'capture_volume', 'charuco', 'checkerboard'] 
            and isinstance(calib[c],dict)]
for cam in cal_keys:
    calib[cam]['size'] = calib[cam]['size'][::-1]

    K = np.array(calib[cam]['matrix'])
    K[0,0], K[1,1] = K[1,1], K[0,0]
    K[0,2], K[1,2] = K[1,2], K[0,2]
    calib[cam]['matrix'] = K.tolist()
    
    r,t = rotate_cam(calib[cam]['rotation'], calib[cam]['translation'], ang_z=-np.radians(rot))
    r = np.squeeze(cv2.Rodrigues(r)[0])
    calib[cam]['rotation'], calib[cam]['translation'] = r.tolist(), t.tolist()

with open(calib_file, 'w') as f:
    toml.dump(calib, f)

```

davidpagnon, 2024-10-07:
I figured out where the problem was!
After all, it was not about handling Caliscope files, nor about distortion (although I should probably tackle it one day or another). The issue was in the script for rotating the calibration file; more specifically, in the optical center `K[0,2], K[1,2]`. 

I edited my above answer with the correction.

JustHere27, 2024-10-08:
I have run the script that you gave, and it correctly orientated the video's
However, I still ran into an error:

```python
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[3], line 4
      2 Pose2Sim.poseEstimation()
      3 Pose2Sim.synchronization()
----> 4 Pose2Sim.triangulation()
      5 Pose2Sim.filtering()
      6 Pose2Sim.markerAugmentation()

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:375, in triangulation(config)
    372 logging.info(f"Project directory: {project_dir}")
    373 logging.info("---------------------------------------------------------------------\n")
--> 375 triangulate_all(config_dict)
    377 end = time.time()
    378 elapsed = end-start

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:900, in triangulate_all(config_dict)
    897 nb_persons_to_detect = len(Q_tot)
    899 if nb_persons_to_detect ==0:
--> 900     raise Exception('No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.')
    902 # IDs of excluded cameras
    903 # id_excluded_cams_tot = [np.concatenate([id_excluded_cams_tot[f][k] for f in range(frames_nb)]) for k in range(keypoints_nb)]
    904 id_excluded_cams_tot = [np.hstack(np.hstack(np.array(id_excluded_cams_tot[n]))) for n in range(nb_persons_to_detect)]

Exception: No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.
```

davidpagnon, 2024-10-08:
Hi, sorry again, there was a problem in the script upon saving the file. I hope it works on your end now!

This is what I get:
[image]

JustHere27, 2024-10-08:
It appears we messed up the setting of the origin point, but other than that it seems to work.
  
https://github.com/user-attachments/assets/5718ba32-c6d7-46a2-a78b-48e08ebaf051

Thanks again for all the help.

davidpagnon, 2024-10-08:
That's interesting 😁 Out of curiosity, do you use the Rajagopal model rather than the Pose2Sim one? If so, what is the reason? (just in case there is an issue I'm not aware of)

You're welcome!

JustHere27, 2024-10-08:
I am not sure if I understand what you mean, however if you are referring to pose_model I left it on default.

closed 2024-10-11 by davidpagnon

mprib, 2025-05-29:
@davidpagnon,

getting pinged with this update filled me with a bit of joy. I'm finalizing my dissertation over the summer and very much looking forward to immersing myself in code and motion capture for a bit when this is all said and done. Hope you are doing well out there!

Mac