perfanalytics/pose2sim #137: wsl rocm documentation

jschoch, 2024-10-17:
I have a bit of an oddball environment and wanted to document issues with it.

firstly you have to get rocm setup and working.  this is a bit off topic but it takes a bit of effort.

Then you can confirm wslg is working by installing somethiinig like gedit and setting the DISPLAY=:0

there is some issue with python-opencv and QT when using "wslg".  I'm not really sure what is happening, but after in installed vlc (a qt app) I was able to run the single person demo.  

in ipython oddly I had to run this QT demo code and then I could run the demo.
```
from PyQt5.QtWidgets import QApplication, QLabel
app = QApplication([])
label = QLabel('Hello World!')
label.show()
app.exec_()

from Pose2Sim import Pose2Sim
Pose2Sim.calibration()
Pose2Sim.poseEstimation()
Pose2Sim.synchronization()
Pose2Sim.personAssociation()
Pose2Sim.triangulation()
Pose2Sim.filtering()
Pose2Sim.markerAugmentation()
Pose2Sim.kinematics()
```

I think there are some ONNX issues falling back to CPU for some things but it runs.

davidpagnon, 2024-10-18:
Interesting, thanks for sharing!

If I may ask, why do you use WSL?
Pose2Sim works on Linux, but it also natively works on Windows, so that would save you a lot of trouble.

jschoch, 2024-10-18:
amd 7900 xtx, only gets rocm in linux :(

davidpagnon, 2024-10-18:
Oh okay, thanks!

jschoch, 2024-10-20:
I also seem to have this problem with sports2d, it actually makes a QT window and then i get the odd error

```
Processing video stream...
100%|█████████████████████████████████████████████████████| 230/230 [01:16<00:00,  3.00it/s]
Video processing completed.
Processed video saved to /home/schoch/sports2d/demo_Sports2D/demo_Sports2D.mp4.
Processed images saved to /home/schoch/sports2d/demo_Sports2D/demo_Sports2D_img.

Post-processing pose:
- Person 0: Interpolating missing sequences if they are smaller than 10 frames. Large gaps filled with last_value.
Filtering with Butterworth filter, 4th order, 3 Hz.
Pose saved to /home/schoch/sports2d/demo_Sports2D/demo_Sports2D_px_person00.trc.
QObject::moveToThread: Current thread (0x431e700) is not the object's thread (0x447ade0).
Cannot move to target thread (0x431e700)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/schoch/miniconda3/envs/sports2d/lib/python3.10/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```

jschoch, 2024-10-20:
crazy long thread on this at stack overflow.  https://stackoverflow.com/questions/46449850/how-to-fix-the-error-qobjectmovetothread-in-opencv-in-python

what i did was:

`
pip uninstall opencv-python
conda install opencv -c conda-forge
`

that seems to fix the qpa plugin issue.  Still digging into to an easy way to run the pose estimation with rocm.

jschoch, 2024-10-23:
rocm works with this change 

```python

@@ -242,6 +242,10 @@ def setup_pose_tracker(det_frequency, mode, tracking):
             device = 'cuda'
             backend = 'onnxruntime'
             logging.info(f"\nValid CUDA installation found: using ONNXRuntime backend with GPU.")
+        elif torch.cuda.is_available() == True and 'ROCMExecutionProvider' in ort.get_available_providers():
+            device = 'rocm'
+            backend = 'onnxruntime'
+            logging.info(f"\nValid CUDA installation found ROCM: using ONNXRuntime backend with GPU.")
         else:
             raise
     except:
```

and this update to rtmlib

https://github.com/Tau-J/rtmlib/issues/38

davidpagnon, 2024-11-03:
Added to the code 👍

closed 2024-11-26 by davidpagnon