perfanalytics/pose2sim #141: Calibration Issue

drcsh7002, 2024-10-29:
when I set both the  ['calibration']['calculate']['extrinsics']['board'] to False, the Pose2Sim will show error,
[image]

davidpagnon, 2024-10-30:
Hi, sorry I'm away from my computer for a few days and I cannot see your screenshot. Can you upload it again?

drcsh7002, 2024-10-30:
[image]

drcsh7002, 2024-10-30:
Thank you for replying to me. I am using your pose2sim to make a small tool that can process my own experimental data. This problem makes me unable to continue

davidpagnon, 2024-10-30:
I'm not sure what you set to False, can you send me a screenshot of the calibration section of your Config.toml file?

drcsh7002, 2024-10-30:
I'm using the Pose2Sim 0.8.4
[image]

davidpagnon, 2024-11-03:
Hi, thanks for the screenshot. I still don't know what the problem is. Can you first update Pose2im with `pip install pose2sim -U`?
You are telling me that setting show_reprojection_error to false leads to an error when you also set calibration_type to 'calculate' and extrinsics_method to "board". Do you get this error with extrinsics_method set to 'scene' (in the Demo trials for example)?

davidpagnon, 2024-11-03:
Okay I could reproduce the error. Let me investigate it

davidpagnon, 2024-11-03:
Okay, I found and fixed the error!
I'll wait for a bit more edits to the code to push a new release, but in the meantime you can do `pip show pose2sim` to find your installation location, look for calibration.py, and edit line 694. Replace 
```python
                imgp, objp = findCorners(img_vid_files[0], extrinsics_corners_nb, objp=object_coords_3d, show=show_reprojection_error)
```
by:

```python
                imgp = findCorners(img_vid_files[0], extrinsics_corners_nb, objp=object_coords_3d, show=show_reprojection_error)
                objp = object_coords_3d
```

davidpagnon, 2024-11-13:
I'm closing it as I believe the issue is fixed (will be added in a future release), but don't hesitate to reopen the issue if needed.

closed 2024-11-13 by davidpagnon