perfanalytics/pose2sim #143: Dealing with nan in ..._filt_<filter>.trc causing condition 'convertOk' not met during kinematics

jtoerber, 2024-11-05:
Hi,
I have 2 Cameras. One of the camera could see my right small toe all of the time, the other not.
So after FIltering I have nan in the trc-File for the corresponding Landmark/Keypoint.
When I do Inverse Kinematics to get the angles, I get an error message:
<pre>
Pose2Sim.kinematics()
  File "C:\Users\toerberj\.conda\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py", line 505, in kinematics
    kinematics(config_dict)
  File "C:\Users\toerberj\.conda\envs\Pose2Sim\lib\site-packages\Pose2Sim\kinematics.py", line 627, in kinematics
    perform_scaling(trc_file, kinematics_dir, osim_setup_dir, model_name, right_left_symmetry=right_left_symmetry, subject_height=subject_height[p], subject_mass=subject_mass[p], remove_scaling_setup=remove_scaling_setup)
  File "C:\Users\toerberj\.conda\envs\Pose2Sim\lib\site-packages\Pose2Sim\kinematics.py", line 476, in perform_scaling
    opensim.ScaleTool(scaling_path_temp).run()
  File "C:\Users\toerberj\.conda\envs\Pose2Sim\lib\opensim\tools.py", line 2220, in __init__
    _tools.ScaleTool_swiginit(self, _tools.new_ScaleTool(*args))
RuntimeError: std::exception in 'OpenSim::ScaleTool::ScaleTool(std::string const &)': SimTK Exception thrown at String.h:466:
  Error detected by Simbody method String::convertTo(): Couldn't interpret string 'inf inf inf' as type T=SimTK::Array_<double,unsigned>.
  (Required condition 'convertOK' was not met.)
</pre>
Lowering the thresholds did not work.
I am not able to work with more than two Cameras as they need high FPS (>=100) in my case.
In this specific case I am able to adjust the positions of the cameras.
But, in another case I want to track arm and fingers and it will happen that there are gaps, which are even larger than (large) interp_if_gap_smaller_than-Parameter, especially for landmarks/keypoints, which are in a specific case not interesting.
How am I able to get InverseKinematics running?
I could replace nan with 0.0?
I could (at least try to) remove the non interesting landmark(s)?
Any recommendations?
Interestingly the filtering has a problem with that not the triangulation, which may decide to take the Landmark/Keypoint from the Camera, that is able to see it.

jtoerber, 2024-11-05:
Issue: #114:  fill_large_gaps_with = 'last_value' # 'last_value', 'nan', or 'zeros' 
Could the (only) value of the other Camera be taken? Or is then in the Case of 2 cameras the 3D information missing?

davidpagnon, 2024-11-05:
Hi, 

You need at least 2 cameras to triangulate the coordinates. When there is a hole for a few frames you can interpolate it, but if a point is not seen for the whole sequence, it is not going to work.

Here are a few ideas, by order of hackiness:
- Capture the sequence again and make sure the foot is seen by both cameras
- Capture with more cameras
- Edit Pose2Sim/OpenSim_Setup/Scaling_Setup_Pose2Sim*.xml, and set <IKMarkerTask name="RSmallToe"> to false
  Do it for your IK_Setup file as well
  Then set `fill_large_gaps_with = 'zeros'` (if it does not work, replace the columns with zeros in pandas or a spreadsheet editor)

Please tell me how it goes!

jtoerber, 2024-11-06:
Hi,
I guess you mean:
```` xml
<IKMarkerTask name="RSmallToe">
    <!--Whether or not this task will be used during inverse kinematics solve, default is true.-->
    <!-- <apply>true</apply> -->
    <apply>false</apply>
    <!--Weight given to the task when solving inverse kinematics problems, default is 0.-->
    <weight>1</weight>
</IKMarkerTask>
```
in 
IK_Setup_Pose2Sim_<skeleton>.xml
Scaling_Setup_Pose2Sim_<skeleton>.xml

With 
fill_large_gaps_with = 'zeros' # 'last_value', 'nan', or 'zeros' 
I get a lot of 0.0, not only for the nan.
With
fill_large_gaps_with = 'last_value' # 'last_value', 'nan', or 'zeros' 
and manually replacing the nan with 0.0 makes "kinematics" possible.

In Opensim there are some issues with it as maybe the number of Cameras may not be enough or the alignment of the two used Cameras is not properly. The person is not really standing nor doing what it did in the Video. In OpenCap there were no problems with two cameras.
(if interested the last_values, zeros and manually are in attached zip-file)
[Demo_SinglePerson.zip](https://github.com/user-attachments/files/17647182/Demo_SinglePerson.zip)

davidpagnon, 2024-11-06:
There are definitely issues in your triangulated trc files. How did you calibrate?

jtoerber, 2024-11-06:
Hi,
I did
Pose2Sim.calibration()
with a 3x5 (45mm) checkerboard.

You did not specify which issues there are in the trc files.
And btw.: The synchronization seems to be not deterministic as I usually get 3 frames difference in this case, but once 166 frames, which are about 10 seconds here. I guess setting a frame-range may help here.

davidpagnon, 2024-11-06:
Check this thread starting from the linked comment maybe. It seems like if you download synchronized videos from OpenCap, they are exported at half the resolution, which leads to a faulty calibration
https://github.com/perfanalytics/pose2sim/issues/142#issuecomment-2454283931

davidpagnon, 2024-11-06:
The issues are that the trc files look don't look like a human being doing anything 😅

Synchronization is likely to be another issue indeed. Check the appropriate doc section to make sure it is correct: https://github.com/perfanalytics/pose2sim/blob/main/README.md#synchronization

> If results are not satisfying, edit your [Config.toml](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/Demo_SinglePerson/Config.toml) file:\
All keypoints can be taken into account, or a subset of them.\
The whole capture can be used for synchronization, or you can choose a time range when the participant is roughly horizontally static but with a clear vertical motion (set `approx_time_maxspeed` and `time_range_around_maxspeed` accordingly). 
> 
> *N.B.:* Works best when:
> - the participant does not move towards or away from the cameras
> - they perform a clear vertical movement
> - the capture lasts at least 5 seconds long, so that there is enough data to synchronize on
> - the capture lasts a few minutes maximum, so that cameras are less likely to [drift with time](https://github.com/mprib/caliscope/discussions/496)

jtoerber, 2024-11-06:
I think in calibration.py you should check the resolution of all the videos and their frames per second.
Your advantage/disadvantage opposite to OpenCap is that you allow all kind of cameras.
This may even lead to inhomogenous cameras or camera settings during one recording.

Got new camera. I'll try my final setting tomorrow.

Synchronization:
I did calibration and poseEstimation. 
Then I did all but calibration and poseEstimation.
Hence same videos and Pose-Estimations and then I got once 166 frames difference instead of 3 frames with a clear right wrist raise in the first seconds and Videos captured in one Thread to be (relatively) safe/close.

davidpagnon, 2024-11-06:
Calibration does check the resolution of videos, I just tried to make a guess based on the information your provided but I went in the wrong direction. To be clearer, if you capture with OpenCap and convert its calibration files, and the export the wrong videos, you will have a discrepancy between the resolutions of the calibration file and the videos. But that's not what you did.

The framerate is also detected automatically, see Config.toml: 
`frame_rate = 'auto' # fps # int or 'auto'. If 'auto', finds from video (or defaults to 60 fps if you work with images) `

Synchronization should be deterministic if you run on the same videos with the same parameters. If the right wrist clearly raises only at one time, it should not give you such different results, unless there is a problem in the parameters of your Config.toml file.

If you want to send some of your data to contact@david-pagnon.com, I can look at it and search for the bug.

jtoerber, 2024-11-08:
What does Pose2Sim.personAssociation() do?
I get:
<pre>
\personAssociation.py:588: RuntimeWarning: Mean of empty slice
  mean_error_px = np.around(np.nanmean(error), decimals=1)

--> Mean reprojection error for Neck point on all frames is nan px, which roughly corresponds to nan mm.
--> In average, 0.01 cameras had to be excluded to reach the demanded 20 px error threshold after excluding points with likelihood below 0.3.

Tracked json files are stored in ...

Associating persons took 00h00m02s.
</pre>

Then when running triangulation I get:
<pre>
...
  raise Exception('No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.')
Exception: No persons have been triangulated. Please check your calibration and your synchronization, or the triangulation parameters in Config.toml.
</pre>

davidpagnon, 2024-11-08:
personAssociation sorts out people to ensure that you triangulate the same person across cameras rather than trying to triangulate two different persons together (in single-person mode, it removes people in the background). If you only have one person in the scene, you do not need to run it. 

This error still means that either your calibration or your synchronization is not good. Did you try trimming the video by hand to make sure the problem was not about synchronization?

jtoerber, 2024-11-08:
Got new Camera. Now complete new setting with two different cameras. New video recording.
Played a little bit with approx_time_maxspeed.

It seems that also personAssociation (and perhaps triangulation) need some minimal video length to work properly (in addition to synchronization). Perhaps you should display a warning or even an error. Switched temporarily to lightweight for my local computer.
With 7 seconds video length it did not work. Now I have 11 seconds.
This worked except kinematics. Skeleton is standing at the beginning, then it is creeping on the ground.

What means 0.03 excluded cameras? 3% of something?

davidpagnon, 2024-11-08:
Hi again, 
It should work with 7 seconds videos, and in any case, kinematics should also work.
It's really hard to help you figure it out, could you maybe send me a project folder (with short videos, I'm running out of space on my laptop), so that I can see what's wrong?

jtoerber, 2024-11-12:
Hi,

in my video cam01.mp4 with the script below I can see that if I use 9 times each 30th frame I get different rvecs values than if I do it only one, two or three times.

If you use
```python
cv2.drawChessboardCorners(img, CHECKERBOARD, corners2, ret)
```
usually the chessboard is red to ... in colors. At some point you can see that the Chessboard is ... to red, so 180 degress rotated or something like that.

Perhaps this helps finding the problem.

Btw.: Could not execute "caliscope" after pip installing it.

How large must the Charuco board be for Freemocap?

[my_calibration.zip](https://github.com/user-attachments/files/17720837/my_calibration.zip)

jtoerber, 2024-11-12:
Hi,

added a little dirty
```python
# to csv; append
    with open('poseEstimation.csv', 'a', newline='') as f:
        writer = csv.writer(f)
        # writer.writerows(keypoints_with_confidence_i)
        writer.writerow(keypoints_with_confidence_i)
```
in poseEstimation.py after line 101. I think we need csv export here as well as in Blazepose_runsave.py.

Usually I would expect that running the pose estimation with "performance" would lead to the best keypoints/landmarks detection.
Than I would look at the *.trc file if there are any nan's. If so I would check the likelihood/confidence/visibility for the landmark in the csv file to decide, if and how to lower the likelihood threshold.

In my personal example only to be able do my tests a threshold of 0.1 and interestingly balance (not performance) lead to no nan's.

davidpagnon, 2024-11-13:
Hi all, sorry my job is keeping me quite busy, but I'll try to answer by the end of the week!

davidpagnon, 2024-11-15:
Hi Jens and Mario,

@medpar, It seems like there is something shady with OpenCap data that I can't figure out there. Would you mind sending me your full project folders, including the calibration files and/or videos?). I suppose there must be some Nan in the TRC file, which would be caused by a bad triangulation. Are results for "1legjumpdroplanding" good?

@jtoerber I just tried the data you are sent me.
- First thing, the toes are not well detected at all. The video is not of great quality but I've never seen anything like that.
  Be aware that COCO_133 models don't perform as well on body points as Halpe_26. So you could either ditch the hand keypoints, or use it in "performance" mode.
  [image]
- Synchronization worked when I set `approx_time_maxspeed = [1.0,1.0]` # the left wrist motion roughly happens at 1 second on each camera
- Considering the poor results of pose estimation, I increased `reproj_error_threshold_triangulation = 30` instead of 15, and `interp_if_gap_smaller_than = 50` instead of 10. Of course, triangulation is not good on toes.
- It seems like the person is lying to the side, which is probably due to the different OpenCap calibration. You can reorient your TRC file prior kinematics with this script: https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/Utilities/trc_Zup_to_Yup.py
- Since the feet are not well detected, I removed them from kinematics. You have to find your Pose2Sim\OpenSim_Setup\Scaling_Setup_Pose2Sim_Coco133.xml and D:\softs\github_david\pose2sim\Pose2Sim\OpenSim_Setup\IK_Setup_Pose2Sim_Coco133.xml files (type `pip show pose2sim` for fine Pose2Sim location). In scaling, set <Measurement name="foot"> to false. In scaling and IK, you should set IKMarkerTask to false for all feet points. I would do the same for the hands, because since they are not well detected, the make the shoulder lock anddisturb the whole body kinematics.
  With so few cameras with poor quality, I would actually tinker with the parameters directly within the OpenSim API: https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#within-opensim-gui

jtoerber, 2024-11-15:
Hi,

there is a misunderstanding. I do not use Opencap. Hence Calibration is done by Pose2Sim.
I need COCO_133 because of fingers.
Currently not much time, so will Take a while to Test it.

And: Would you please so kind and remove the Image in your last Post. Thanks in advance.

davidpagnon, 2024-11-17:
Sure, sorry for the image!

Okay, I may have misunderstood concerning OpenCap then. In this case, you just need to flip Y and Z coordinates and make Z=-Z in your configuration file and you should have the person upright!

So by order of impact, you should:
- have better lighting
- use more cameras
- use the performance mode

davidpagnon, 2024-12-01:
This issue has been marked as stale. Do not hesitate to reopen it if needed!

jtoerber, 2024-12-08:
Hi,
with the settings
```
reproj_error_threshold_triangulation = 30
interp_if_gap_smaller_than = 50
```
(and some transformation of coordinates) I can see my capturing in OpenSim, but:
The upper body first is correct and then rotates 180 degrees, so that the knees show to the front and the head looks back.

When using OpenCap we had no problem with 2 cameras. So does the Pose2Sim triangulation need at least 3 cameras to work properly?
For me it is additionally important, because I only have 2 cameras with high FPS rate and am not able to add more cameras of those.
(From other videos I myself now generated a trc-file with Mediapipe matching your Pose2Sim model, which works in OpenSim)

jtoerber, 2024-12-08:
@davidpagnon There is no Reopen issue button so it seems I have not the right to do so.

davidpagnon, 2024-12-09:
Hi, 
2 cameras are of course not ideal, but we have made it work even in multi-person mode and challenging conditions (such as handstand, large occlusions, etc), so it should definitely work.
Did you try on performance mode and with better lighting? From what I remember there were a lot of flickering where the feet were detected on the head, and probably some right-left switches.

jtoerber, 2024-12-15:
Hi,
was recording new Video with 120 FPS.

During Pose-Estimation the following error occurs:

[mpeg4 @ 0000018e88722b80] timebase 1000/119193 not supported by MPEG 4 standard, the maximum admitted value for the timebase denominator is 65535
[ERROR:0@3.917] global cap_ffmpeg_impl.hpp:3194 open Could not open codec mpeg4, error: Unspecified error (-22)
[ERROR:0@3.917] global cap_ffmpeg_impl.hpp:3211 open VIDEOIO/FFMPEG: Failed to initialize VideoWriter

In [project] I can set frame_rate to an int instead of 'auto'.

davidpagnon, 2024-12-16:
Hi, sorry for the late answer, I'm pretty swamped with work this week... I'll do my best to spend some time on issues by the end of the week!

jtoerber, 2024-12-17:
Hi,

tried raw RTMPose (COCO_133) with leightweight and performance. In my CSV-File I now have (original) keypoint-Column-Names via:
```python
def getInfos(keypoints, openpose_skeleton):
    # https://github.com/Tau-J/rtmlib/blob/main/rtmlib/visualization/draw.py
    num_keypoints = keypoints.shape[1]
    if openpose_skeleton:
        if num_keypoints == 18:
            skeleton = 'openpose18'
        elif num_keypoints == 134:
            skeleton = 'openpose134'
        elif num_keypoints == 26:
            skeleton = 'halpe26'
        else:
            raise NotImplementedError
    else:
        if num_keypoints == 17:
            skeleton = 'coco17'
        elif num_keypoints == 133:
            skeleton = 'coco133'
        elif num_keypoints == 21:
            skeleton = 'hand21'
        elif num_keypoints == 26:
            skeleton = 'halpe26'
        else:
            raise NotImplementedError

    skeleton_dict = eval(f'{skeleton}')
    keypoint_info = skeleton_dict['keypoint_info']
    skeleton_info = skeleton_dict['skeleton_info']
    
    return keypoint_info, skeleton_info

def getKeypointNamesAndIds(keypoints, Coord2DPlusScore=False, openpose_skeleton=True):
    keypoint_info, skeleton_info = getInfos(keypoints, openpose_skeleton)
    keypointNames = []
    keypointIds = []
    for i, kpt_info in keypoint_info.items():
        if Coord2DPlusScore:
            keypointNames.append(kpt_info['name'] + "_x")
            keypointNames.append(kpt_info['name'] + "_y")
            keypointNames.append(kpt_info['name'] + "_score")
            
            keypointIds.append(str(kpt_info['id']) + "_x")
            keypointIds.append(str(kpt_info['id']) + "_y")
            keypointIds.append(str(kpt_info['id']) + "_score")
        else:
            keypointNames.append(kpt_info['name'])
            keypointIds.append(kpt_info['id'])
    
    return keypointNames, keypointIds
```

min(score) for lightweight was about 0,24 (, which btw. is not bad regarding only one camera was used from left behind)
min(score) for performance was about 2,4

If no person is in the frame, this is detected (resized by namedWindow):
[image]

If a person is detected, the sport devices (see above) are not detected.

So:
- scores are not in [0,1] at least for performance (<-> confidence, visivility ?)
- region of interest detection and/or skeleton detection has some problems if no person is in the frame

Sounds a little strange

davidpagnon, 2024-12-21:
> With the limitation to only accept horizontal checkerboard patterns it would even get worse with a checkerborard laying on the ground.

This is why I do not recommend using a checkerboard for extrinsic parameters, but rather coordinates from objects in the scene. Then, as long as they are not symmetric, you can have cameras looking in any direction (see the appropriate section in the documentation: https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#calculate-from-scratch).

> But in my understanding the underlying human skeleton model should not allow feet pointing to the front and shoulder/hips pointing to the back.

It should not allow it indeed, did this happen? Do you have a model and a mot file for me to check? (if the scaling is particularly wrong, you could have very weird things though)

> [mpeg4 @ 0000018e88722b80] timebase 1000/119193 not supported by MPEG 4 standard, the maximum admitted value for the timebase denominator is 65535
[ERROR:0@3.917] global cap_ffmpeg_impl.hpp:3194 open Could not open codec mpeg4, error: Unspecified error (-22)
[ERROR:0@3.917] global cap_ffmpeg_impl.hpp:3211 open VIDEOIO/FFMPEG: Failed to initialize VideoWriter

Are you telling me that pose estimation fails at 120 fps? At what point does it fail? Can you (generally) give me the full error message? Did you manage to solve it?
It seems like the detected framerate is 119.193, which is strange. If you can directly edit in the code L179 and L380 `cap.get(cv2.CAP_PROP_FPS)` to `round(cap.get(cv2.CAP_PROP_FPS))`, it should take care of it. I'll push this edit to the next release.

Concerning the confidence score that is sometimes higher than 1, it is surprising but they do let it be on purpose, see there: https://github.com/open-mmlab/mmpose/issues/884
And I did notice, too, that sometimes RTMPose had false positives.

jtoerber, 2024-12-23:
Hi,
regarding FPS:
No, it is not failing, but you get the above error messages. My cameras were set to 120 FPS. So yes you should round that to avoid the messages.

Regarding Skeleton detection in the "dark" videos:
Setting: pose_model = 'COCO_133' (hence rtmlib and to_openpose=False)
In this context setting "performance" does the work. And yes, one should use better lightning.
I could observe the following:
I added bounding boxes and Person<index>-Text to the pose estimation step.
1. With setting "lightning":
In some frames I can see 2 Person-Bounding-Boxes although there is only one person!
2. With setting "balanced":
In some frames I can see some weird lines, but only one Person-Bounding-Box, perhaps due to the same problem "lightweight" has.
In this case I would say the model should not allow that.
3. With setting "performance":
Only one Person-Bounding-Box detected, no weird lines.

Perhaps problem in 2. leads to the 180 degree turn of the upper body. Will check that.

Regarding scores:
Do you do something with the scores or do you compare your "likelihood_threshold" directly with the scores?
Then at least a hint in the Config.toml file for mode "performance" would be nice.

I would also add the Bounding Boxes that you show in Synchronization step during Multi-Person in Pose Estimation step, at least optional, because of the above reason. It helps finding weird effects.
Would also be interesting if the "second person detection" has influences on JSON-Output with setting "lightweight".

jtoerber, 2024-12-25:
Hi,
reagrding tracking of persons/sort_people_sports2d(...):
I would expect that the pose estimation framework does the tracking. This seems to be a non-trivial problem:
https://ethz.ch/content/dam/ethz/special-interest/baug/igp/photogrammetry-remote-sensing-dam/documents/pdf/[HenLeaRosGCPR2014.pdf](https://ethz.ch/content/dam/ethz/special-interest/baug/igp/photogrammetry-remote-sensing-dam/documents/pdf/HenLeaRosGCPR2014.pdf):
...
The problem of multiple people tracking is related to many computer vision
applications. A common way to tackle the problem is to divide it into two sub-problems: (i) the detection of all objects and (ii) the correct aggregation of these
detections along time to form the final trajectories. People detectors [3, 7] achieve
high accuracy if the scene is not too crowded. Wrong detections occur especially
when people are partially occluded or when they are standing too close to each
other, being interpreted as one person.
The data association framework links detections in time, either on a frame-by-frame basis (online systems) or by taking longer sequences into account (offline systems). Recent advances have been made by formulating the problem
of finding the trajectories for all objects and all frames. Solutions are obtained
by solving a combinatorial optimization problem.
...
Prof. Rosenhahn told us that Min-Cut-Optimization can be used here (I think for the bounding rectangles). Occlusion is a big problem. He is working on that.

davidpagnon, 2025-01-03:
> 2\. In some frames I can see some weird lines, but only one Person-Bounding-Box, perhaps due to the same problem "lightweight" has.

The weird lines are due to some points detected in completely wrong places. I've seen that on your video but there must be something that puts the network off because I've never seen it anywhere else. 

> Do you do something with the scores or do you compare your "likelihood_threshold" directly with the scores?

The triangulation is weighted by the scores, which improves 3D results. There is also a threshold on scores (likelihood) below which a point is not taken into consideration in the synchronization, person association, and triangulation stages.

> Would also be interesting if the "second person detection" has influences on JSON-Output with setting "lightweight".

Do you mean, when several people are detected on the same person? This is the issue with the lightweight mode: there are often false positives. It could be solved by using another person detection model. I plan to work on giving more flexibility to the user this weekend (if I find the time), so that they can choose their own detection and/or pose model.

> I would expect that the pose estimation framework does the tracking.

I have tried different approaches for tracking:
- the one proposed by RTMlib, which is not great. I have recoded a simple Intersection over Union tracker, which is slightly better but often not sufficient
- My own tracking algorithm (included in Pose2Sim and Sports2D) works better and is enough in most cases
- In the most challenging cases, crowded scenes in particular, I have used [DeepSortRealtime](https://github.com/levan92/deep_sort_realtime/tree/master), which is more robust but slower, requires some (simple) parametrization, and still fails in the most difficult situations.

closed 2025-01-21 by davidpagnon

jtoerber, 2025-02-01:
Hi,

it seems that the Joint Angle range in the osim file is only for the GUI, but not for inverse kinematics.

Found this articles regarding number of Cameras (2 vs. 3, 4) and Position of the Cameras:
https://www.sciencedirect.com/science/article/pii/S0021929025000454?dgcid=rss_sd_all

Found some hints in the Internet that intrinsic calibration (or undistortion afterwards) may help improving reprojection error. Maybe that's why OpenCap works with 2 Cameras and restricts the supported Smartphones.

davidpagnon, 2025-02-02:
> it seems that the Joint Angle range in the osim file is only for the GUI, but not for inverse kinematics.

Inverse kinematics can be used both in the GUI or in command line, and joint angle ranges should be respected in both cases. Or did I understand it wrong?

I saw this article in my google scholar feed, too. It was interesting, thanks!
I definitely need to work on the keypoint undistortion as well. It should not be complicated but I struggle to find the time.

jtoerber, 2025-03-25:
Hi,

if inverse kinematics only needs one trc-file row to calculate the output angles mot-file-row then Interpolation or even Extrapolation on angle level would be good (with respect to the constraints), because then the sphere on which the arms of the joint(s) operate can't be left. So it can't become unphysiological except for velocity or accelerations.
It seems that the InverseKinematics-API holds a kind of state or kind of a session, because the opensim.log can't be deleted in between runs.
Seems a little bit the wrong direction, but could help to cope with too large interpolation gaps.

davidpagnon, 2025-03-26:
> It seems that the InverseKinematics-API holds a kind of state or kind of a session, because the opensim.log can't be deleted in between runs.

I did notice this but I'm not sure how to fix this or if it is possible. If you find a way, I'm interested!

> if inverse kinematics only needs one trc-file row to calculate the output angles mot-file-row then Interpolation or even Extrapolation on angle level would be good (with respect to the constraints), [...]

I'm not quite sure I understand this. If gaps are already interpolatedat the trc level, there should not be any gap at the ik levek, right? "because then the sphere on which the arms of the joint(s) operate can't be left" This is the part I do not get.

jtoerber, 2025-04-28:
polar coordinates: r(cos(alpha), sin(alpha)). Interpolation on Basis of alpha(t) would then be on the S2 sphere, e.g. for elbow moving around shoulderjoint or wrist around elbowjoint.

It seems that OpenSim uses 3D Butterworth filter during inverse kinematics, so you may have to address this.

[ShoulderElbowAngle.pdf](https://github.com/user-attachments/files/19938426/ShoulderElbowAngle.pdf)