perfanalytics/pose2sim #144: Core dump on camera sync, qt issue

kaangoksal, 2024-11-07:
Hi, I followed the instructions to install using conda, however when I'm doing syncronization, I get core dump 

```bash
(Pose2Sim) ubuntu@circlePC:~/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson$ ipython
Python 3.9.20 (main, Oct  3 2024, 07:27:41) 
Type 'copyright', 'credits' or 'license' for more information
IPython 8.18.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from Pose2Sim import Pose2Sim
   ...: Pose2Sim.calibration()
   ...: Pose2Sim.poseEstimation()
   ...: Pose2Sim.synchronization()

---------------------------------------------------------------------
Camera calibration
On Wednesday 06. November 2024, 22:58:28
Calibration directory: /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson/calibration
---------------------------------------------------------------------

Converting /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson/calibration/Calib.qca.txt to .toml calibration file...

--> Residual (RMS) calibration errors for each camera are respectively [0.221, 0.235, 0.171, 0.191] px, 
which corresponds to [0.402, 0.445, 0.45, 0.505] mm.

Calibration file is stored at /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson/calibration/Calib_qualisys.toml.

Calibration took 0.00 s.

---------------------------------------------------------------------
Pose estimation for Demo_SinglePerson, for all frames.
On Wednesday 06. November 2024, 22:58:28
Project directory: /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson
---------------------------------------------------------------------

Valid CUDA installation found: using ONNXRuntime backend with GPU.
Inference run on every single frame.
Using HALPE_26 model (body and feet) for pose estimation.
Mode: balanced.

2024-11-06 22:58:30.440522389 [W:onnxruntime:, session_state.cc:1166 VerifyEachNodeIsAssignedToAnEp] Some nodes were not assigned to the preferred execution providers which may or may not have an negative impact on performance. e.g. ORT explicitly assigns shape related ops to CPU to improve perf.
2024-11-06 22:58:30.440557004 [W:onnxruntime:, session_state.cc:1168 VerifyEachNodeIsAssignedToAnEp] Rerunning with verbose output on a non-minimal build will show node assignments.
load /home/ubuntu/.cache/rtmlib/hub/checkpoints/yolox_m_8xb8-300e_humanart-c2c7a14a.onnx with onnxruntime backend
2024-11-06 22:58:30.595352413 [W:onnxruntime:, session_state.cc:1166 VerifyEachNodeIsAssignedToAnEp] Some nodes were not assigned to the preferred execution providers which may or may not have an negative impact on performance. e.g. ORT explicitly assigns shape related ops to CPU to improve perf.
2024-11-06 22:58:30.595373212 [W:onnxruntime:, session_state.cc:1168 VerifyEachNodeIsAssignedToAnEp] Rerunning with verbose output on a non-minimal build will show node assignments.
load /home/ubuntu/.cache/rtmlib/hub/checkpoints/rtmpose-m_simcc-body7_pt-body7-halpe26_700e-256x192-4d3e73dd_20230605.onnx with onnxruntime backend

Estimating pose...
Found video files with extension mp4.
Processing cam01.mp4: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 100/100 [00:13<00:00,  7.60it/s]
--> Output video saved to /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson/pose/cam01_pose.mp4.
Processing cam03.mp4: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 100/100 [00:05<00:00, 16.91it/s]
--> Output video saved to /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson/pose/cam03_pose.mp4.
Processing cam02.mp4: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 100/100 [00:07<00:00, 13.74it/s]
--> Output video saved to /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson/pose/cam02_pose.mp4.
Processing cam04.mp4: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 100/100 [00:06<00:00, 15.41it/s]
--> Output video saved to /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson/pose/cam04_pose.mp4.

Pose estimation took 00h00m35s.

---------------------------------------------------------------------
Camera synchronization
On Wednesday 06. November 2024, 22:59:04
Project directory: /home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/Pose2Sim/Demo_SinglePerson
---------------------------------------------------------------------

Synchronization is calculated on the whole sequence. This may take a while.
Keypoints used to compute the best synchronization offset: ['RWrist'].
These keypoints are filtered with a Butterworth filter (cut-off frequency: 6 Hz, order: 4).
They are removed when their likelihood is below 0.4.

Synchronizing...
QObject::moveToThread: Current thread (0x1e15270) is not the object's thread (0xc21be00).
Cannot move to target thread (0x1e15270)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, linuxfb, minimal, offscreen, vnc, webgl.

Aborted (core dumped)
```

Did anyone encounter this before?

davidpagnon, 2024-11-07:
Hi, I think you are running on a headless server, aren't you? If so, you need to deactivate all the display parameters in Config.toml (display-detection, display_sync_plots, display_figures).

I generally recommend carefully thinking about the synchronization configuration when you run headless, because a bad synchronization will lead to incoherent triangulation results.

davidpagnon, 2024-11-07:
If I misdiagnosed the problem and you run Linux with a screen, I am not sure. Can you try and Uninstaller and reinstall opencv?

```python
pip uninstall opencv-python opencv-python-headless
pip install opencv-python
```

Otherwise, can you copy-paste your error in chatGPT and try what they suggest? Tell me where it leads you if you need it.

kaangoksal, 2024-11-07:
Hey, thanks for fast reply. I'm on a ubuntu desktop 22.04, I have two screens and two nvidia rtx 3060s. 

I'm now trying with this 

```python

import os
import sys
from PyQt5 import QtCore

os.environ["QT_QPA_PLATFORM_PLUGIN_PATH"] = "/home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/cv2/qt/plugins"
os.environ["QT_QPA_PLATFORM"] = "offscreen"

from Pose2Sim import Pose2Sim
Pose2Sim.calibration()
Pose2Sim.poseEstimation()
Pose2Sim.synchronization()
Pose2Sim.personAssociation()
Pose2Sim.triangulation()
Pose2Sim.filtering()
Pose2Sim.markerAugmentation()
Pose2Sim.kinematics()

```

this lead to this 

```bash
(Pose2Sim) ubuntu@circlePC:~/Desktop/pose2sim/Pose2Sim/Demo_SinglePerson$ ipython
Python 3.9.20 (main, Oct  3 2024, 07:27:41) 
Type 'copyright', 'credits' or 'license' for more information
IPython 8.18.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import os
   ...: import sys
   ...: from PyQt5 import QtCore
   ...: 
   ...: os.environ["QT_QPA_PLATFORM_PLUGIN_PATH"] = "/home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.9/site-packages/cv2/qt/plugins"
   ...: os.environ["QT_QPA_PLATFORM"] = "offscreen"
   ...: 
   ...: 
   ...: from Pose2Sim import Pose2Sim
   ...: Pose2Sim.calibration()
   ...: Pose2Sim.poseEstimation()
   ...: Pose2Sim.synchronization()

---------------------------------------------------------------------
Camera calibration
On Wednesday 06. November 2024, 23:12:20
Calibration directory: /home/ubuntu/Desktop/pose2sim/Pose2Sim/Demo_SinglePerson/calibration
---------------------------------------------------------------------

Converting /home/ubuntu/Desktop/pose2sim/Pose2Sim/Demo_SinglePerson/calibration/Calib.qca.txt to .toml calibration file...

--> Residual (RMS) calibration errors for each camera are respectively [0.221, 0.235, 0.171, 0.191] px, 
which corresponds to [0.402, 0.445, 0.45, 0.505] mm.

Calibration file is stored at /home/ubuntu/Desktop/pose2sim/Pose2Sim/Demo_SinglePerson/calibration/Calib_qualisys.toml.

Calibration took 0.00 s.

---------------------------------------------------------------------
Pose estimation for Demo_SinglePerson, for all frames.
On Wednesday 06. November 2024, 23:12:20
Project directory: /home/ubuntu/Desktop/pose2sim/Pose2Sim/Demo_SinglePerson
---------------------------------------------------------------------

No valid CUDA installation found: using OpenVINO backend with CPU.
Inference run on every single frame.
Using HALPE_26 model (body and feet) for pose estimation.
Mode: balanced.

load /home/ubuntu/.cache/rtmlib/hub/checkpoints/yolox_m_8xb8-300e_humanart-c2c7a14a.onnx with openvino backend
load /home/ubuntu/.cache/rtmlib/hub/checkpoints/rtmpose-m_simcc-body7_pt-body7-halpe26_700e-256x192-4d3e73dd_20230605.onnx with openvino backend

Estimating pose...
Skipping pose estimation as it has already been done. Set overwrite_pose to true in Config.toml if you want to run it again.

Pose estimation took 00h00m00s.

---------------------------------------------------------------------
Camera synchronization
On Wednesday 06. November 2024, 23:12:21
Project directory: /home/ubuntu/Desktop/pose2sim/Pose2Sim/Demo_SinglePerson
---------------------------------------------------------------------

Synchronization is calculated on the whole sequence. This may take a while.
Keypoints used to compute the best synchronization offset: ['RWrist'].
These keypoints are filtered with a Butterworth filter (cut-off frequency: 6 Hz, order: 4).
They are removed when their likelihood is below 0.4.

Synchronizing...
This plugin does not support propagateSizeHints()
This plugin does not support raise()

```

I think it got stuck at Camera synchronization. I don't see much cpu action.

kaangoksal, 2024-11-13:
Everything worked after I turned off the `display_sync_plots`

closed 2024-11-13 by kaangoksal