perfanalytics/pose2sim #150: The marker points of the model do not match the marker points of the TRC file

smwscth, 2024-11-26:
Hello，I successfully reproduced the action in Opensim using my own recorded image, but when I want to execute IK with a scaled model, I find that the marker point of the trc file does not match the scaled model, what is the cause of this? Thank you for your answer

smwscth, 2024-11-26:
[image]

davidpagnon, 2024-11-26:
Oh okay, thank you for reporting this! 

This is due to our last change on the LSTM model, made to handle wrist motion when the wholebody pose model is run. But we overlooked the fact that we need separate model and setup files depending on which pose model we rum.

@hunminkim98, do you think you could do that while you are working on other aspects of the OpenSim files? 

In the meantime, just untick the read lines and it should work. I'll keep you updated when we release a new working version, hopefully today!

hunminkim98, 2024-11-26:
Hello,
Sorry for the confusion caused by the last update.
I tested it using the command Pose2Sim.kinematics(), and it worked properly even when hand keypoints were not present.
However, it should, of course, work in the OpenSim GUI as well. I’ll work on that!
The LSTM models will be:
- Model_Pose2Sim_withoutHands_LSTM.osim
- Model_Pose2Sim_withHands_LSTM.osim

Have a great day!

davidpagnon, 2024-12-01:
It apparently works fine when running scaling and IK within Pose2Sim, but you need to untick the parameters when doing it in the GUI. In any case, it should be fixed now, thanks to HunMin.
Please tell me how it goes on your side @smwscth

smwscth, 2024-12-01:
> It apparently works fine when running scaling and IK within Pose2Sim, but you need to untick the parameters when doing it in the GUI. In any case, it should be fixed now, thanks to HunMin. Please tell me how it goes on your side @smwscth

I'll test if it's working as soon as possible, thanks for your work，have a great day!

smwscth, 2024-12-06:
Hello, I updated to the new version, but when I used the new video test, I reported an error in the last step, may I ask what caused it? 
And when I used the trc file I got for IK, the result was very strange, I recorded a standing motion, but the model was lying on the ground.
[image]

hunminkim98, 2024-12-06:
> I'll test if it's working as soon as possible, thanks for your work，have a great day!

Hello, I think the issue might be caused by the 'inf' value when running Pose2Sim.kinematics(). The 'inf' value could originate from improper data in your .trc file or an issue in the scaling setup file. However, I believe the setup file should be fine since I tested it several times, and it worked correctly with pytest.

Did you use the same pose estimation model and LSTM for this new test? Also, did the preview of the .trc file before marker augmentation look good in the OpenSim GUI?

hunminkim98, 2024-12-06:
> Hello, I updated to the new version, but when I used the new video test, I reported an error in the last step, may I ask what caused it? And when I used the trc file I got for IK, the result was [...]

Hello, I think the issue might be caused by the 'inf' value when running Pose2Sim.kinematics(). The 'inf' value could originate from improper data in your .trc file or an issue in the scaling setup file. However, I believe the setup file should be fine since I tested it several times, and it worked correctly with pytest.

Did you use the same pose estimation model and LSTM for this new test? Also, did the preview of the .trc file before marker augmentation look good in the OpenSim GUI?

davidpagnon, 2024-12-06:
Thanks HunMin for participating.
Also @smwscth, did you check whether your trc file (in the pose-3d folder) looked good? If not, the problem may arise from a bad calibration

smwscth, 2024-12-08:
> Thanks HunMin for participating. Also @smwscth, did you check whether your trc file (in the pose-3d folder) looked good? If not, the problem may arise from a bad calibration.
> 
> Edit: sorry HunMin, you already asked the question about the trc file!
Hello, I check my trc file, and I'm sure it's because my trc file isn't good, but I don't know why my calibration is bad, I think these errors are aceptable
[image]

smwscth, 2024-12-08:
@davidpagnon Is this due to poor detection posture? I found out that RTM would detect the robotic arm as a person, and there was a problem with the detection of my foot. Thanks for your answer
[image]
[image]

davidpagnon, 2024-12-08:
Hi, some oublier détections should not be a problem. However,  how many videos do you have? And can you see the whole person at all times for at least 2 videos?

If not and you are only interested in the upper body, edit the custom model in config.toml and remove all the lower body points (and choose it as your "pose_model")

smwscth, 2024-12-08:
Hello, I shot a new set of videos, using 3 cameras to test, I confirmed that the calibration results are good, and the RTM test results are not bad, but I reported an error in the triangulation this time. I was confused because the first time I tried it I got good results, but the next few times I repeated the process I got errors. Could you please help me see what is wrong with my project files?
[image]
[image]

smwscth, 2024-12-09:
@davidpagnon I know what went wrong. I changed the camera position later, but I copied the original Calib_qualisys, Calib_scene file. When I re-calibrated the scene, Calib_qualisys would not write the new one. Once I removed Calib_qualisys, it worked fine, thanks for your work

davidpagnon, 2024-12-09:
Makes sense! I'm glad it works now. 
I'm closing the issue, but feel free to comment to reopen it if needed.

closed 2024-12-09 by davidpagnon

chengyuZou, 2025-03-19:
> Hello，I successfully reproduced the action in Opensim using my own recorded image, but when I want to execute IK with a scaled model, I find that the marker point of the trc file does not match the scaled model, what is the cause of this? Thank you for your answer

Hello, I heard that you successfully reproduced the motion in OpenSim using your own recorded images. Could you please teach me how to do this? Thank you!

davidpagnon, 2025-03-19:
Hi @chengyuZou,
Can you open a new issue and be a little more specific? At what stage of the pipeline do you start to see something strange? Don't hesitate to attach a screenshot or two :)