perfanalytics/pose2sim #153: Fingers in skeletons.py and osim-File (OpenSim)

jtoerber, 2024-12-09:
Hi,

you mentioned a ToDo regarding fingers.
I can see that only Pinky, Index and Thumb are used/mapped.
In OpenSim they are additionally locked, so we do not really get angles.
I Would Like to have the wrist Angle. There is an article by Hao Ni et al from June 2024 on arxiv, that announces a Python Module calculating ISB conform Joint angles. The wirst Angle is calculated via the Thumb.
There are articles which use the middle finger for the wrist Angle. Currently this one is not mapped.
Why are the fingers locked in OpenSim?
Is there a Problem in OpenSim?

I could also see a (dead) Link on simtk-Website regarding fingers.

davidpagnon, 2024-12-09:
Hi,

The wrist angles are unlocked in the full-body OpenSim models (body135, coco133, Halpe68_136.xml, withHands_LSTM), and they are determined by the Thumb,  Index, and Pinky markers. 

It is possible to add a fully featured hand model to the Pose2Sim model with all finger motions, but I am not sure when I can do it.

What dead link are you talking about?

davidpagnon, 2024-12-11:
You are right, it might be worth releasing the range of motion!

jtoerber, 2024-12-25:
Hi,
regarding wrist angle:
Config.toml-setting:
pose_model = 'COCO_133'
If use_augmentation = true in [kinematics]
Model_Pose2Sim_LSTM.osim
is taken, where wrist flex is locked, not
Model_Pose2Sim_Coco133.osim

I do not need the augmentation and know the problem of this KI extension (for specifiy movements).

Besides: I cannot see "withHands_LSTM" anywhere. Currently not important for me.

Ok, thumb, index and pinky span a plane. I could also see articles that take the middle finger. Some use Euler angles, some Quaternions. It seems that wrist angle is not really defined and/or the handling of the exceptions.

Btw.: The SMPL Topology also has a marker on the lower spine column:
https://dl.acm.org/doi/pdf/10.1145/2816795.2818013
(volumetric approach instead of skeleton)
Then no augmentation for this marker is needed.

closed 2025-01-21 by davidpagnon