perfanalytics/pose2sim #155: Failed to connect Socket 'child_frame' of type PhysicalFrame

chengyuZou, 2025-01-19:
Hi
I got the .mot, .trc, and .osim files from "Demonstration Part-1: End to end video to 3D joint angle computation". However, when I try to open the .osim file in OpenSim, I encounter the following error:

Failed to connect Socket 'child_frame' of type PhysicalFrame (details: Model 'Pose2Sim_scaled'::findComponent() cannot find a nameless subcomponent.).
In Object 'ground_pelvis' of type CustomJoint.
Thrown at Component.cpp:287 in finalizeConnections().

How can I resolve this issue? Thank you.

davidpagnon, 2025-01-20:
Hi, 

It should work so I'm not sure what the issue is, but let's try and find the cause of it.

Which version do you use (`pip show pose2sim`) and how did you install it ([option 1 or 2](https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#installation))? 

Does it fail during the `Pose2Sim.kinematics()` stage or only when you try to open the model in OpenSim? Which version of OpenSim? Which OS?

What is the name of the .osim file you are trying to open? 
I suspect a problem in the Geometry files. Can you try to copy-paste all the files from `<pose2sim_install_folder>/Pose2Sim/OpenSim_Setup/Geometry` to the Geometry folder you will find in OpenSim GUI -> Edit -> Preferences -> Geometry search path?

davidpagnon, 2025-01-20:
Potentially helpful: Try to remove the oldest OpenSim version and to reinstall the newest one:
https://github.com/opensim-org/opensim-core/issues/2206#issuecomment-430419758

chengyuZou, 2025-01-21:
Got it! The reason was that the OpenSim version was not the latest
Thanks!

closed 2025-01-21 by chengyuZou