perfanalytics/pose2sim #157: Unable to use GPU to perform the tasks

LeJeanDeMarley, 2025-01-22:
Hi,

I'm currently working on a scientific research project that requires me to use the GPU of my computer in order to perform a large amount of tasks on multiple subjects. 

After reading and trying my best to follow the instructions found in **_Installation / 4. Optionnal_**, and try to debug if necessary, i'm currently stuck on this error message that i can't quite figure out as a beginner in this field : _See Image1_

Current informations 
- computer : 
- installation versions : 
 - CUDA : 12.7 _See Image2_
 - ONNX Runtime : 1.20.x
 - cuDNN : 9.x
 - PyTorch : 2.5.0 from : conda install pytorch==2.5.0 torchvision==0.20.0 torchaudio==2.5.0 pytorch-cuda=12.4 -c pytorch -c nvidia

Image1 : screenshot from the anaconda terminal window with the error message 
Image2 : screenshot from the _nvidia-smi_ cmd window

Any help or clue on how to solve this issue would be greatly appreciated. Thank you to the team for the amazing tool that Pose2Sim is. 

Best regards. 

[image]
[image]

davidpagnon, 2025-01-30:
Hi @LeJeanDeMarley,
Sorry I missed this issue! I just had a child birth and was not on top of the game. 
Out of curiosity, how did you solve it?

LeJeanDeMarley, 2025-01-30:
Hello David,

First of all congratuliations to you and your family, I hope that everyone
is happy and healthy !

I currently work on a study in Lyon in partnership with Trinoma and one of
their engineer showed me the command " *nvidia-smi dmon -s u*" which helped
me to have a better understanding of the GPU usage, which was working
perfectly. Previously i only checked the task manager ressources which was
indicating 0% usage of the GPU.

I still have this message at the beginning of the pose estimation phase,
but as i tried the "lightweight" "balanced" and "performance" modes, the
GPU is in fact used as needed. I will work toward making sure the GPU is
used at it's full capacity if possible, since in balanced mode it only used
21-53% of it's capacity, as it was shown in the terminal.
If needed i can keep you updated on this issue, and I just also realised i
could've written this mail in french as well.

Best regards,

Yanis Marciot.

davidpagnon, 2025-01-30:
Perfect then! And if you find out how to make your GPU work up to 100%, I'm interested as I do not have an answer currently.
Et merci 😊