perfanalytics/pose2sim #158: Faster intrinsics camera calibration ?

ANaaim, 2025-02-04:
Hello, 

For some cases i have VERY long intrinsics calculations. 

https://github.com/perfanalytics/pose2sim/blob/51c88e6ebe09bee3a47ab6f68156c56d73cb6b18/Pose2Sim/calibration.py#L612C1-L613C109
````python
        ret_cam, mtx, dist, rvecs, tvecs = cv2.calibrateCamera(objpoints, imgpoints, img.shape[1::-1], 
                                    None, None, flags=(cv2.CALIB_FIX_K3))# + cv2.CALIB_FIX_PRINCIPAL_POINT))
````
to accelerate it it could become : 
````python
        ret_cam, mtx, dist, rvecs, tvecs = cv2.calibrateCamera(objpoints, imgpoints, img.shape[1::-1], 
                                    None, None, flags=(cv2.CALIB_FIX_K3+cv2.CALIB_USE_LU))# + cv2.CALIB_FIX_PRINCIPAL_POINT))
````
as suggested below it give similar results (should maybe be investigated in more detail)

https://stackoverflow.com/questions/51073309/why-does-the-camera-calibration-in-opencv-python-takes-more-than-30-minutes

Any thought about it ? 

Best regards

davidpagnon, 2025-02-05:
I would have suggested calibrating with fewer images, but I must say that this approach sounds pretty convincing!
I even looked into triangulation using LU decomposition rather than SVD, but it does not seem relevant in this case.

Have you tried it yourself? If so and it looks convincing in practice, feel free to submit a pull request :)

ANaaim, 2025-02-05:
Maybe I am trying to have to much different point of view from my checkerboard. I tried it my self, it seems alright. 

I could check if we have huge difference in my case and if not I can do a pull request ?

davidpagnon, 2025-02-05:
I would appreciate it, thanks in advance :)
From what I can see in various sources, I expect it to be good, too.

davidpagnon, 2025-02-09:
I just pushed it with another release on a bunch of various things but it was probably a bit optimistic. If you ever see anything wrong, please tell me and I'll revert back to the previous version!