perfanalytics/pose2sim #162: Regarding Camera Synchronization Functionality Using a Flashlight

pattyhao, 2025-03-04:
Hi，

Thank you for your assistance with the following inquiry:
I encountered an issue. In the README, it is mentioned, "N.B.: Alternatively, synchronize cameras using a flashlight, a clap, or a clear event. GoPro cameras can also be synchronized with a timecode, by GPS (outdoors), or with their app (slightly less reliable)." 
However, in synchronization.py, I only found functionality for synchronization based on vertical motion. We experimented with using a flashlight, but the results were not satisfactory. 
Could you please clarify how to use the flashlight for camera synchronization? Or do we need to implement this feature ourselves?

davidpagnon, 2025-03-04:
Hi! There is no script for that, unfortunately. What people usually do is extract all frames, check when the light goes on, and remove all frames so that all cameras start at the same time.

What problem do you face when running the synchronization based on vertical speed?

pattyhao, 2025-03-04:
OK, I see...
Synchronization based on vertical speed runs very well!
Thank you for your help~

closed 2025-03-04 by davidpagnon