perfanalytics/pose2sim #169: [BUG] In calibration l.644 os.walk does not enforce a sorted list.

ANaaim, 2025-03-22:
```python
        extrinsics_cam_listdirs_names = next(os.walk(os.path.join(calib_dir, 'extrinsics')))[1]
```

It should be : 
```python
        extrinsics_cam_listdirs_names = sorted(next(os.walk(os.path.join(calib_dir, 'extrinsics')))[1])
```

I can do a PR (but not now ==> I had a problem Two time in the last month and I forget about it) as a result a bad intrinsic matrix can be mixed with the video.

(i suppose a similar modification should be done )

Maybe something more elegant should be done 
such as : 
take the name of the camera processed, 
look at the position in the list of this specific camera 
and use this indices instead of the i in : 

```python
for i,cam in enumerate(extrinsics_cam_listdirs_names
....
mtx,dist ) np.array(K[i], np.array(D[i])

```

(sorry for the basic message I am in a rush for processing data... but i dont want to forget this)

Bye :D

davidpagnon, 2025-03-22:
Thanks, I wonder if this would explain some weird results a user had a while ago... https://github.com/perfanalytics/pose2sim/issues/146

If you are up for submitting a pull request I'm happy with it! No rush.

ANaaim, 2025-03-23:
It could explain effectively... as the user regenerate data maybe, by luck, the data were correctly associated with each camera...

closed 2025-04-01 by davidpagnon