perfanalytics/pose2sim #171: Unable to have a good result

Paispe02, 2025-04-01:
Hi people

I've been trying for a long long time to have a good result and after a lot of tries, I don't really know what else to do. I have solved a lot of problems that have been arising. First, I tried to do this with two phones, and that didn't worked, the errors where so High that all the time some errors arose and ir didn't work. Know I have recorded this with 4 cameras and it gives me no error while executing, but the kinematics result is horrible and makes no sense. I also have to do the calibration mannually, as if not the person goes under the floor. 

Here I leave a video of the result that I get on opensim and the zip where all the videos and code are. If some one could help me solve my errors would be awesome.

https://upcomillas-my.sharepoint.com/:u:/g/personal/202001616_alu_comillas_edu/EUFgEUu7mOBKoAOI-ngIsysByCtadMVqfDncO-LVvGTOfQ?e=YmE0Bz

https://github.com/user-attachments/assets/d63f69c5-5731-417a-ae68-58dcfbb2f385

davidpagnon, 2025-04-01:
Hi,

The good news is, you are not far off! I think that the main problem is met at the synchronization stage.

Here are a few tips:
- If you want to debug something, it is much easier to work with small files and videos. You will not have to wait for ages before you know whether it worked or not, so you can try something else quickly

**Calibration:** 
- If you do not want to calibrate manually, the only other option for now is to convert another calibration file. But the one in the Demo folders would not work since it corresponds to completely different cameras, set in a place that probably has nothing to do with yours  (we are working on automatic calibration, but we are not there yet)
- You can remove the board after calibration, so that you are sure you do not risk slipping on it while dancing and jumping
- Obviously, the board you use for extrinsic calibration is not ideal as it consists in paper sheeds that are approximately taped together, but it is enough to have something working, even if not with the best accuracy. 
- I generally recommend using scene measurements instead of a board.  It will be more accurate and would prevent you from having the occasional flip upside down. You could measure a few distances based on the planks on your wooden floor. You can use tape to remember where to click

**Pose estimation:**
- Nothing particular 

**Synchronization:**
- You are doing the same movement repeatitively, which makes it hard to know whether a cycle is the first, second, or so on. The tool could find out that the best correlation is between the second cycle of the first camera and the third of the second one, for example.
- If you want to understand how things work, I would recommend activating `synchronization_gui = true` and `display_sync_plots = true` (`pip install pose2sim -U` to upgrade to the last version)
- It seems like your videos were mostly synchronized already, how did you do? If so, you do not have to run synchronization.

**Person association:**
- No need to run it since there is one single person in the scene

**Triangulation and kinematics:**
- No particular issues on my side
- I did not run filtering nor marker augmentation but you would probably just get smoother results

https://github.com/user-attachments/assets/628910a2-6553-4694-9c69-07ba2424cd21

[logs.txt](https://github.com/user-attachments/files/19558166/logs.txt)

davidpagnon, 2025-04-02:
You are welcome :)

- Concerning calibration with scene measurements, you can just take measurements in your room, put the meter coordinates in the config file, and then click on the points in the right order. Tell me if it is still not clear!
- Concerning the synchronization GUI, you need to install the latest Pose2Sim version (`pip install pose2sim -U`) and look at the Config.toml file you are using. [Here](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/Demo_SinglePerson/Config.toml) is the latest version. 
- So there are two options: either you synchronize your videos with DaVinci resolve, then you don't need to do it via Pose2Sim; or you do not use Davinci resolve, and then you synchronize your pose estimation results with Pose2Sim
- Actually I may have spoken a bit fast, 20 seconds like in your videos is rather good for synchronization. Too short will make it less robust.
- I saw your reflection in the mirror, but it is rather small so I think this will always be the second detection, which is ignored if you do not run person association. But person association should not hinder your results anyway.
- It should work with 2 cameras; it will be less stable, especially if the calibration is not great, but I've had good results in the past.

I have deleted your folder and do not have much time to try it again, but 
- I think the main issue comes from synchronization. I'm not sure why you did not have similar results, but since you already synchronized your videos, there should be close to 0 frames of offset and one of your cameras had 40 or so. 
- My calibration results were slightly better than your, but probably not enough to explain such differences
- I would be surprised if it were a cause, but I did not run person association, so something wrong may have happened there
- Also, you may want to clean up your project: only keep your videos, pose, and calibration directories, as well as your Config.toml file. Make sure you have the right number of video directories in each folder, with names in the same order. In case the calibration instrinsics were wrong the first time, it is better to always recalculate them by setting `overwrite_intrinsics=true` in the Config.toml file

Paispe02, 2025-04-20:
To close this Issue. I've been trying again and again and couldn't achieve a good result. 

I decided to try something different and do it again in my desk computer. It has better specs and it runs on windows instead of my mac. I installed Pose2Sim and run the code and it was perfect and smooth.

I tryed to install everything again in my Mac to give it a fresh start and with the same folder and the same files I had a completely different and deviated result that did not worked well. 

Currently I have a MacBook Pro M1 with 16Gb of Ram. I don't know it there is a problem with the processing of my mac or with the OS which does not really make sense, but I was unable to get a result in my Mac. 

I can't give you any insight of what is the problem as I don't have it detected and I'll continue working on my desk computer. Thanks a lot for the help.

davidpagnon, 2025-04-21:
Interesting, I wonder what the problem is... I test the code [on an M1 processor](https://github.com/perfanalytics/pose2sim/actions/runs/14535421301/job/40782752820), too, and results are good (I also know at least one person for whom it works). Do you know at which stage it starts to diverge?
What's your Python version (`python -V`)? What's your Pose2Sim version (`pip show pose2sim`)?

Anyway, I'm glad it works on your Windows computer, at least!

Paispe02, 2025-05-14:
Hi again. I'm having some trouble again in my windows this time. I'm using now other videos with a better calibration board and I get an error in triangulation that does not allow me to continue. I've detected some problems, while running the script. The videos are syncronized by an external trigger, but still the program detects an offset when doing the syncronization. I've tried also to execute without the syncronization but still does not work. I also did the person association to try solve this issue, and I got some strange results also, it excluded 1.31 cameras. Were can be the problem? I can expeculate with the size of the images, or that the images are not easy to work as the angles are bad but I can't understand why it just blocks me from continuing instead of just giving me a bad result. I've also tryed deleting one camera and trying also with two but still gives me the same error.

If someone had the same problem or knows what I'm doing wrong would be really helpful.
https://upcomillas-my.sharepoint.com/:f:/g/personal/202001616_alu_comillas_edu/Ej-vGwYRL3BCsmYuf7hBA8AB3iwRRYFauK8Tc6t-SnZJcw?e=5UFNtz

[image]

davidpagnon, 2025-05-15:
Hi, I had a quick look, and unfortunately, the way the cameras are laid out makes it quite challenging to triangulate coordinates.

Two of them are almost parallel to each other, and one is vertically above the subject. It is generally advised for the cameras to be at a 30-45° angles regarding each other.

I have not gone further for now, as I'm swamped with work, but this should be the main issue. Good luck with everything!

Paispe02, 2025-05-17:
Yep, I imagine it was that, just needed to see if it was another problem or the cameras. Thank you!

closed 2025-05-17 by Paispe02