perfanalytics/pose2sim #172: Have problems using my data totriangulate keypoints.

WOWOWO222, 2025-04-17:
Hello, David. I've been trying to extract kinematics from my own videos. I used two phones to finish this work. I used two phones to record my videos. However , when I'm trying to triangulate keypoints, I met errors follow.

[image]

To solve this problem, should I add more cameras or calibrate my cameras again?

davidpagnon, 2025-04-17:
Hi, the first thing to check is your calibration. How did you calibrate? What is the magnitude of the error? Were the reprojected points correctly overlaid to the clicked points?

WOWOWO222, 2025-04-18:
> Hi, the first thing to check is your calibration. How did you calibrate? What is the magnitude of the error? Were the reprojected points correctly overlaid to the clicked points?

Hi, sorry for my late reply. I used checkboard to do calibrate the extrinsics . However, the result looks terrible. What should I do to use checkboard to finish the calibration of extrinsics? 

[image]

WOWOWO222, 2025-04-20:
> Okay, so the issue definitely comes from your calibration.
> 
> * **Intrinsics:** Did you use the same checkerboard? How many images did you use? Note that you cannot use the intrinsic calibration from the Demo folders, as you probably do not use the exact same cameras.
> * **Extrinsics:** I suppose the image you attached is a crop of your original image? If not, I do not see where your participants would have enough space to move in such a small space. Note that the extrinsic calibration is made to determine where your cameras are, so once it is done, you cannot move them anymore.
> 
> This section of the tutorials might help you: https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#calculate-from-scratch

Thanks, David. I 've retried the calibration and when triangulate I encountered this error. I wonder how did it happen and what should I do to fix it?

[image]

[image]

davidpagnon, 2025-04-20:
I've never seen this error yet 🤔
Did the calibration go well? What was the error?

WOWOWO222, 2025-04-21:
> I've never seen this error yet 🤔 Did the calibration go well? What was the error?

WOWOWO222, 2025-04-21:
> I've never seen this error yet 🤔 Did the calibration go well? What was the error?

Hi, David. I worked out how this happened. When I edited the videos, they were compressed and do not match my cameras. Meanwhile, the motion of the person in the videos did not match from time to time. So I reedited them and solved the problem.  However , when I was trying to extract motions from my videos ,I met this error. I can only get the .osim file, but without the .mot. And I've tried many times to fix it but it did not work. How could I solve it? 

[image]

[image]

davidpagnon, 2025-04-21:
There is a problem at some point, probably at the triangulation stage or earlier. Could you send me your directory so that I can figure it out? dp2032@bath.ac.uk

davidpagnon, 2025-04-24:
Solved by email. Feel free to reopen the issue if needed!

closed 2025-04-24 by davidpagnon