perfanalytics/pose2sim #175: Applied to RTSP real-time network cameras

MoonTTMM, 2025-04-23:
I successfully use 3 cameras to record mp4 files, and process files to get .mot file. It work fine and can be played in opensim GUI. Thanks a lot.
 
Now I am trying to apply it to RTSP real-time network cameras, is there any sample code for this?
Or any suggestions to use the this project in real-time camera way?

AurelienCoppee, 2025-04-23:
Hello,
I'm currently reworking the Pose2Sim pipeline to enable real-time frame processing. As part of this, I'm implementing a system for selecting input sources. I haven't yet considered RTSP integration, but if you have some time, I'd be happy to walk you through how it could be added.

MoonTTMM, 2025-04-24:
Hi, 
I am really looking forward to the real-time frame processing mode. I’d also appreciate your guidance on integrating RTSP cameras.

MoonTTMM, 2025-04-28:
Hi @AurelienCoppee ,  when would the real-time frame processing feature be launched?

AurelienCoppee, 2025-04-28:
I'm quite busy right now, but I hope as soon as possible, in 3 weeks maybe...