perfanalytics/pose2sim #177: extrinsics Error

chengyuZou, 2025-04-28:
Hi David!
When performing extrinsic calibration using a checkerboard, the points on Cam1 are displayed correctly, but the points on Cam2 are severely misaligned. Due to this misalignment, an error occurred during the Pose2Sim.markerAugmentation() step . How can I fix this? Thanks

Cam01:
[image]

Cam02:
[image]

My_Config:
`extrinsics_method = 'scene'`
`object_coords_3d =   [[0.0 , 0.0 , 0.0],
                              [0.0 , 0.048 , 0.0],
                              [0.0 , 0.096 , 0.0],
                              [0.0 , 0.144 , 0.0],
                              [0.0 , 0.0 , -0.048],
                              [0.0 , 0.048 , -0.048],
                              [0.0 , 0.096 , -0.048],
                              [0.0 , 0.144 , -0.048],
                              [0.0 , 0.0 , -0.096],
                              [0.0 , 0.048 , -0.096],
                              [0.0 , 0.096 , -0.096],
                              [0.0 , 0.144 , -0.096],
                              [0.0 , 0.0 , -0.144],
                              [0.0 , 0.048 , -0.144],
                              [0.0 , 0.096 , -0.144],
                              [0.0 , 0.144 , -0.144],
                              [0.0 , 0.0 , -0.192],
                              [0.0 , 0.048 , -0.192],
                              [0.0 , 0.096 , -0.192],
                              [0.0 , 0.144 , -0.192],
                              [0.0 , 0.0 , -0.240],
                              [0.0 , 0.048 , -0.240],
                              [0.0 , 0.096 , -0.240],
                              [0.0 , 0.144 , -0.240],
                              [0.0 , 0.0 , -0.288],
                              [0.0 , 0.048 , -0.288],
                              [0.0 , 0.096 , -0.288],
                              [0.0 , 0.144 , -0.288],
                              [0.0 , 0.0 , -0.336],
                              [0.0 , 0.048 , -0.336],
                              [0.0 , 0.096 , -0.336],
                              [0.0 , 0.144 , -0.336]]`

Log:

[logs.txt](https://github.com/user-attachments/files/19941333/logs.txt)

davidpagnon, 2025-04-28:
Hi @chengyuZou, 
You're the second or third with this issue, could you send me your calibrarion folder and your configuration file? I haven't figured out why this occasionally happened yet.
Contact@david-pagnon.com 

(Of course, I'll use it only for debugging purposes)

chengyuZou, 2025-04-29:
Folder:
https://drive.google.com/file/d/1ltdwQVMW9hoTcseEvvASLy9O7bSWdAmn/view?usp=sharing

I see it! I am trying to find out why this happened too. Thanks for your help

chengyuZou, 2025-04-29:
Hi David, I changed the shooting position, which resolved the extrinsic parameters issue. However, new problems have arisen —errors are now occurring in the final two steps  Pose2Sim.markerAugmentation()  and Pose2Sim.kinematics()
What might be the cause of this? Thank you.

Log:

[logs.txt](https://github.com/user-attachments/files/19961749/logs.txt)

davidpagnon, 2025-04-29:
Hi, I'm not sure I'm gonna have time to check this before the end of the week, but I won't forget!

davidpagnon, 2025-04-30:
@chengyuZou So what di you do, exactly, for the calibration to work? It's still very unclear to me, so I'd be interested in debugging it once for all.

Can you send me your folder once again so that I can see why marker augmentation failed?

chengyuZou, 2025-04-30:
Calibration Work
I simultaneously changed the positions of both the checkerboard and the cameras, and recalibrated under the same conditions. The result was successful. 

As for the final two steps, I'm not certain. 

Here is my file.
https://drive.google.com/file/d/163h1J6SY10Fvlnf5xAbMbNZVzr9GOEEJ/view?usp=drive_link

Thanks David!

davidpagnon, 2025-05-02:
Hi @chengyuZou, I requested access to your drive, but did not get it yet :)

chengyuZou, 2025-05-02:
Is this ok ?
https://drive.google.com/file/d/163h1J6SY10Fvlnf5xAbMbNZVzr9GOEEJ/view?usp=drive_link

davidpagnon, 2025-05-02:
It still asks me to request access, which I do. You should get an email asking you to confirm :)

chengyuZou, 2025-05-02:
Sorry,I have never used Goole mail and Drive before.
maybe this time is ok

davidpagnon, 2025-05-02:
Okay, got it!
I'll have a look in a bit

davidpagnon, 2025-05-02:
Okay, so I did not go all the way, but there are a few issues with your calibration. Feel free to carefully read this section of the [Readme.md file](https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#calculate-from-scratch:~:text=A%20common%20error%20is%20to%20specify%20the%20external%2C%20instead%20of%20the%20internal%20number%20of%20corners%20(one%20less%20than%20the%20count%20from%20calib.io).%20This%20may%20be%20one%20less%20than%20you%20would%20intuitively%20think.)
- First of all, you need to edit your Config.toml file to suit your configuration. In particular, your checkerboard had 7*10 corners, with 15 mm square size 
- The extension will need to be changed, too. 
- And it would be better to have more images for the intrinsic calibration
- In general, if your calibration errors are over 20 mm, you should probably just do it again, because everything else will be impacted.

Important, too, I strongly advise you to set your GoPro in "linear" mode, as strong distorsion will have an impact on your results (so, do not use wide, hyperview, and similar modes)

chengyuZou, 2025-05-03:
Got it ! I will try it again. Thanks David!

chengyuZou, 2025-05-03:
Hi David! I did it again. And Calibration Error is more smaller than before. 
Residual (RMS) calibration errors for each camera are respectively [3.001, 2.95] px, which corresponds to [3.48, 3.532] mm.
However I countered  two situation 

1. the same markerAugmentation Error
Error 
ufunc 'divide' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''

2. I can run Pose2Sim.kinematics() . Getting .osim and .mot files . However they are unusual(height ratio and motion)

Error
https://github.com/user-attachments/assets/37e5db87-e8d2-4204-a1bb-83c433a6509d

My_log
[logs.txt](https://github.com/user-attachments/files/20022506/logs.txt)

My_Drvie
https://drive.google.com/file/d/18UFAOLHWXCTyDX1M6qQgnaeYWGk6NjS4/view?usp=drive_link

Thanks!

davidpagnon, 2025-05-04:
Here are my next remarks:
- I do not think you set the GoPro field of view as Linear, could you change it?
- Your image format is different for the intrinsic parameters, the extrinsic ones, and the actual capture. The properties of the camera change if you change the settings, so you need to take videos with the exact same parameters in all three stages
- The extrinsic points you chose are too close to each other. You do not need to use a checkerboard, you could measure out corners on the wall and on the floor if it is easier.

Also, if there is one single person in the scene, you do not need to run personAssociation()

chengyuZou, 2025-05-04:
Got it . I will try it later and close the issue until other problems happen
Thanks!

closed 2025-05-04 by chengyuZou