perfanalytics/pose2sim #178: Inaccurate 3D reconstruction for squats despite accurate calibration and sync – walking is fine

ValentineUniversiteParisSaclay, 2025-04-29:
Hello,
I am currently using Pose2Sim to reconstruct 3D motion from a multi-camera setup. Although the pipeline runs without crashing, the resulting 3D skeleton is not accurate. Here is a summary of my setup and observations:

**1. Setup**
Number of cameras: 3 Ipad (Ipad 7)
Video synchronization: Audio clap used for precise alignment

**2. Calibration details**
Intrinsic calibration: Done using a checkerboard, following OpenCV standards
Extrinsic calibration: 
Performed manually using reference 3D points visible in all camera
These 3D coordinates were obtained from a Qualisys system 
2D projections were manually clicked in each view for Pose2Sim calibration

**3. Problem**
The 3D reconstruction works reasonably well for walking movements, with consistent joint positions and plausible kinematics.
However, squat movements are reconstructed very poorly : 
- The body appears distorted, with joint positions clearly incorrect
- Lower limb trajectories are inconsistent and anatomically implausible
This inconsistency is surprising, especially given that both walking and squats were captured in the same video with the same setup, same calibration, and same synchronization method. 

**4. What I tried**
Checked that OpenPose 2D detections look accurate for both movements
Re-reviewed calibration files for consistency and correct axis orientation

Thanks in advance for your help,

Best regards,
DUPONT Valentine

davidpagnon, 2025-04-29:
Hi, I'm not sure I'm gonna have time to check this before the end of the week, but I won't forget!

ValentineUniversiteParisSaclay, 2025-04-30:
Thank you for your response. I am attaching all of my data.

[(https://drive.google.com/drive/folders/1incL3DQE493mma2U7rDrISki8dVUWKcE?usp=drive_link)]

ValentineUniversiteParisSaclay, 2025-04-30:
Hello, I think I solved the problem by performing a longer intrinsic calibration with more images and by replacing the cameras with GoPros. 

However, the processing still stops after the triangulation step, and I don’t get access to the filtering or the kinematics stage. Do you know why this might be happening?

davidpagnon, 2025-04-30:
Okay, thanks for your data!

Let's check it stage by stage:

1. **Calibration:** This is where your first issue may arise. I would expect an error of about 2 cm or less, and here is what I obtain:
   `--> Residual (RMS) calibration errors for each camera are respectively [25.845, 10.53, 18.356] px,
which corresponds to [59.23, 36.494, 74.704] mm.`
   This could either come from a poor intrinsic calibration (your checkerboard stays in the middle of the frame instead of going everywhere), or from inaccurate 3D points in the extrinsic stage (but I believe you got the coordinates with markers?)
2. **Synchronization:** No particular issue, but have you tried the synchronization tool proposed by Pose2Sim? It may be easier than using an audio clap, and it worked on my end. I set `synchronization_gui` and `display_sync_plots` to true, and chose the shoulder and head points (which had the clearest vertical motion) between 5.5 s +/- 2s.
3. **Person association:** I did not run person association, since you are the only one in the scene.
 
 
In spite of this, triangulation should run smoothly, and it does not. So there is definitely something wrong with your calibration.

---

I got it!
I'm not sure how you did it, but the videos used for intrinsic calibration had a 720x1280 resolution, while the ones for extrinsic calibration and triangulation had a 1080x1920 resolution.  If the resolution is different, the intrinsic parameters will change, and then everything will be messed up.

Please tell me how it goes when you have a chance!

ValentineUniversiteParisSaclay, 2025-04-30:
Hello, thank you for your help! I will try these suggestions and get back to you with the results.

ValentineUniversiteParisSaclay, 2025-05-02:
Hello, thank you again for your help! The triangulation results are very clean and much more realistic. However, once the triangulation is completed, the following message appears, and neither the filtering nor the augmentation is applied.

[image]

davidpagnon, 2025-05-02:
That's half a good news! I have no idea where this error comes from, I've never heard of it.

Have you tried one of these methods?
- Upgrading numpy, or changing the environment variable: https://stackoverflow.com/questions/64209238/error-15-initializing-libiomp5md-dll-but-found-libiomp5md-dll-already-initial
- Deleting libiomp5md.dll: https://github.com/alievk/avatarify-python/issues/452

davidpagnon, 2025-05-15:
This issue did not get any answers in over 2 weeks, it will now be closed.
Feel free to reopen it if needed!

closed 2025-05-15 by davidpagnon