perfanalytics/pose2sim #181: Pose model download not working

mohamedyousuf1, 2025-05-07:
Hi, while walking through the installation steps. Igot the following error after running the following step:

```
from Pose2Sim import Pose2Sim
Pose2Sim.calibration()
Pose2Sim.poseEstimation()
Pose2Sim.synchronization()
Pose2Sim.personAssociation()
Pose2Sim.triangulation()
Pose2Sim.filtering()
Pose2Sim.markerAugmentation()
Pose2Sim.kinematics()
```

here is the error:

Inference run only every 4 frames. Inbetween, pose estimation tracks previously detected points.

Estimating pose...
Using HALPE_26 model (body and feet) for pose estimation.
```

Valid CUDA installation found: using ONNXRuntime backend with GPU.
Downloading: "https://download.openmmlab.com/mmpose/v1/projects/rtmposev1/onnx_sdk/yolox_m_8xb8-300e_humanart-c2c7a14a.zip" to C:\Users\nowtr\.cache\rtmlib\hub\checkpoints\yolox_m_8xb8-300e_humanart-c2c7a14a.zip
Error: Pose estimation failed. Check in Config.toml that pose_model and mode are valid.
Traceback (most recent call last):
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\poseEstimation.py", line 572, in estimate_pose_all     
    pose_listdirs_names = next(os.walk(pose_dir))[1]
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\urllib\request.py", line 1348, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\http\client.py", line 1283, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\http\client.py", line 1329, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\http\client.py", line 1278, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\http\client.py", line 1038, in _send_output
    self.send(msg)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\http\client.py", line 976, in send
    self.connect()
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\http\client.py", line 1448, in connect
    super().connect()
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\http\client.py", line 942, in connect
    self.sock = self._create_connection(
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\socket.py", line 836, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\socket.py", line 967, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno 11001] getaddrinfo failed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\poseEstimation.py", line 583, in estimate_pose_all     
    pose_tracker = setup_pose_tracker(ModelClass, det_frequency, mode, False, backend, device)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\poseEstimation.py", line 86, in setup_pose_tracker     
    pose_tracker = PoseTracker(
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\rtmlib\tools\solution\pose_tracker.py", line 149, in __init__   
    model = solution(mode=mode,
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\rtmlib\tools\solution\body_with_feet.py", line 104, in __init__ 
    self.det_model = YOLOX(det,
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\rtmlib\tools\object_detection\yolox.py", line 20, in __init__   
    super().__init__(onnx_model,
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\rtmlib\tools\base.py", line 43, in __init__
    onnx_model = download_checkpoint(onnx_model)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\rtmlib\tools\file.py", line 144, in download_checkpoint
    download_url_to_file(url, cached_file, hash_prefix, progress=progress)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\rtmlib\tools\file.py", line 47, in download_url_to_file
    u = urlopen(req)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\urllib\request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\urllib\request.py", line 519, in open
    response = self._open(req, data)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\urllib\request.py", line 536, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\urllib\request.py", line 496, in _call_chain
    result = func(*args)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\urllib\request.py", line 1391, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno 11001] getaddrinfo failed>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py", line 401, in runAll
    pipeline.runAll(**kwargs)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py", line 297, in runAll
    self.poseEstimation()
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py", line 216, in poseEstimation
    estimate_pose_all(config_dict)
  File "C:\Users\nowtr\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\poseEstimation.py", line 586, in estimate_pose_all
    raise ValueError('Error: Pose estimation failed. Check in Config.toml that pose_model and mode are valid.')
ValueError: Error: Pose estimation failed. Check in Config.toml that pose_model and mode are valid.
```

It seems like the 4 links in the config file are not working. Any help would be appreciated.

davidpagnon, 2025-05-07:
Hi,  before I start digging into the bugs, is the link in the second line of your error message not working for you? It works fine on my end 🤔

mohamedyousuf1, 2025-05-07:
Thank you. Maybe the university blocked the website, so changing the internet made it work.
THank you!

closed 2025-05-07 by mohamedyousuf1