perfanalytics/pose2sim #185: kinematics bad results

chengyuZou, 2025-05-12:
Hi David 
I'm encountering issues during the kinematics step, specifically:

- The initial total squared error is very large.
- In the generated .osim file, the feet appear disproportionately large.
- When importing the .mot file, the resulting motion does not match my actual movements.

Details:
Each brick is a square with a side length of 80 cm.
--> Residual (RMS) calibration errors for each camera are respectively [7.875, 9.699] px, 
which corresponds to [6.47, 8.205] mm.

My_motion
https://github.com/user-attachments/assets/b284ef83-66d6-4bea-9537-4d6c05aa824c

OpenSim_motion
https://github.com/user-attachments/assets/bccee614-cdef-4c99-a071-e675e554f53c

Do you know what is going on? Thanks!

Here are my files,all imformation inside
https://drive.google.com/file/d/1VuLUxE2X4bLRBGOdUkMT-5YnVZp2XI1S/view?usp=drive_link

chengyuZou, 2025-05-14:
I know the reason
calibration images are not in the same format as the videos before,that is why the error occur
Now it is more precise  and total squared error is between 0.1 and 0.04

https://github.com/user-attachments/assets/6d72bf11-996d-4b22-9401-02d1f6f276db

And David . Would you like me to contribute? I can write a tutorial blog on how I used my own data from calibration to generating my motion with images and videos

davidpagnon, 2025-05-15:
Nice! The knees are doing weird things at some point, but at least you've got something. Tell me again, how many cameras are you using?

And of course, I'm always up for contributions :)

chengyuZou, 2025-05-16:
Two GoPro12  
Parallel placement

davidpagnon, 2025-05-16:
Did you set them in linear field of view?
It is also better to have them oriented at an angle about 30 to 60 degrees.

Generally, the main problem is calibration. Once this is solved, everything goes more smoothly.

chengyuZou, 2025-05-16:
Linear：yes
Cam placement：
[image]

davidpagnon, 2025-05-16:
Okay! I would put them at an angle towards each other (30 to 60 degrees), it would make triangulation easier.

chengyuZou, 2025-05-17:
Hi David 
My tutorial is here.
https://github.com/chengyuZou/Pose2Sim-Tutorial
I didn't know how to contribute with fork(because tutorial is not code), so I created a repository myself

closed 2025-05-17 by chengyuZou