perfanalytics/pose2sim #186: File Not Found, falling at the first hurdle

Heard-of-Checkerboard, 2025-05-20:
Hello all,

I'm trying to use my own data from DeepLabCut of a hand moving over a desk as a test for future work with birds. I'm pretty sure I've set up the necessary files, including sourcing a calibration TOML from AniPose and translating the h5 pose files. However, I can't even get started because I keep getting a "FileNotFound" error with the Config file when I attempt to calibrate or  #perform pose estimation. I've gone through the motions of determining that I'm in the right directory and that the files are in that directory, but the error won't go away. What step did I miss to dictate the use of my custom Config file? 

I'm sure this is a really simple issue, but I'm not a programmer and I've been using Python for less than a month. 

(Also my sincerest apologies to David for emailing earlier, there's really no excuse for that.)

davidpagnon, 2025-05-20:
Hi, no problem!

I answerd by email but I'll copy-paste it here in case it helps other people:

For Pose2Sim to work, you need a directory structure with at least a Configuration file, a calibration folder, and a video folder.
You can copy-paste the directories wherever you want, and replace the files with the proper ones. You will probably need to change the calibration parameters of your configuration file (convert -> calculate, and edit the dimensions of your checkerboard and scene points).
Then, you need to open your Anaconda prompt in the right folder. Do `cd <PATH_TO_YOUR_FOLDER`, and then open your python command and follow the tutorial.

I hope it helps!

Heard-of-Checkerboard, 2025-05-20:
Hi David,
Thank you for responding on both channels, I appreciate that. 

I've set up my folders as you suggested, but I'm still getting this error. What's interesting, if I replace the Config file from one of the demo folders with my own, I get the same error again. If I switch it back the error goes away. It's as if the system can't recognise my own file at all. 

I'm also trying to work with custom DeepLabCut and OpenSim models, but I feel like that is unrelated for now. 

Would it be useful to attach my file for you to test?

Thank you for the help so far, hopefully we can work through this!

davidpagnon, 2025-05-20:
Can ypu first send me your full error log?

Heard-of-Checkerboard, 2025-05-20:
Sure, it's

`In[1]:    from Pose2Sim import Pose2Sim
 In[2]:    Pose2Sim.poseEstimation()
---------------------------------------------------------------------------------------------------
FileNotFoundError                                                                Traceback (most recent call last)
File ~\minoconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:135, in read_config_files(config)
      133 try:
      134      #if batch
--> 135      session_config_dict = toml.load(os.path.join(config_dir, '..','Config.toml'))
      136      trial_config_dict = toml.load(os.path.join(config_dir, 'Config.toml'))

File ~\miniconda3\envs\Pose2Sim\lib\site-packages-toml\decoder.py:133, in load(f, _dict, decoder)
      132 if_ispath(f):
--> 133     with io.open(_getpath(f), econding='utf-8') as ffile:
      134              return loads(ffile.read(), _dict, decoder)

FileNotFoundError: [Errno 2} No such file or directory: '.\\..\\Config.toml'

During the handling of the above exception, another exception occured:

TomlDecodeError                                                                   Traceback (most recent call last)
Cell In[2], line 1
----> 1 Pose2Sim.poseEstimation()

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:372, in poseEstimation(config)
       371 def poseEstimation(config=None):
--> 372        pipeline = Pose2SimPipeline(config)
       373       pipeline.poseEstimation()

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:163, in Pose2SimPipeline.__init__(self, config)
       162 def __init__(self, config=None):
-->  163      self.level, self.config_dicts = read_config_files(config)
       164       try:
       165            self.session_dir = os.path.realpath([os.getcwd() if self.level==2 else os.path.jion(os.getcwd(), '..')][0])

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:140, in read_config_files(config)
       137      session_config_dict = recursive_update(session_config_dict,trial_config_dict)
       138 except:
       139         # if single trial
-->  140         session_config_dict = toml.load(os.path.join(config_dir, 'Config.toml'))
       141 session_config_dict.get("project").update({"project_dir":config})
       142 config_dicts = [session_config_dict]

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\toml\decoder.py:134, in load(f, _dict, decoder)
       132 if _ispath(f):
       133      with io.open(_getpath(f), encoding='utf-8') as ffile:
-->  134            return loads(ffile.read(), _dict, decoder)
       135 elif isinstance(f, list):
       136       from os import path as op

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\toml\decoder.py:340, in loads(s, _dict, decoder)
       338 if openstring or multilinestr:
       339        if not multilinestr:
-->  340              raise TomlDecodeError("unbalanced quotes", original, i)
       341        if ((sl[i - 1] == "'" or sl[i - 1] == '"') and (
       342                  sl[i - 2] == sl[i - 1])):
       343              sl[i] = sl[i - 1]

TomlDecodeError: Unbalanced quotes (line 277 column 18 char 17187)`

At least that's what it looks like when I copy-paste it here. Part of this I recognise as my own syntax error in writing the new Config file for my own data, but the first bit I'm clueless.

davidpagnon, 2025-05-20:
Are you sure you are running from the right folder? Can you do `pwd` when you are in the python command line, and tell me where your custom directory is?

By the way, if you ran and converted files from DeepLabCut, no need to run pose estimation. Same if you calibrated from anoposelib. And same if you have a single person, natively synchronized cameras, etc. You might just need to run Pose2Sim.triangulation().

Note that you can also try to run hand detection natively from Pose2Sim. You will have to set `pose_model = Hand` and `mode=lightweight`. Not sure it is good enough, but I'm curious to hear how it goes!
(you could also retrain the model in balanced or performance mode with MMPose, but I'm not sure you are ready for this yet)

Heard-of-Checkerboard, 2025-05-20:
After running `pwd` the output was 
`C:\\Users\\andjs\\miniconda3\\envs\\Pose2Sim\\Lib\\site-packages\\Pose2Sim\\Demo_handtest`
which is my custom directory. 

You're probably right that I'm okay skipping ahead to triangulation with how much back-end stuff I've run, but I wanted to test the system to see if it was working the way it was supposed to. Also the error persisted when I tried that, so I'm not sure I've left square one yet. 

I'll try running my footage with `pose_model = Hand` for now just to make sure there isn't something completely wrong behind the scenes, but I'm going to eventually need to run footage of birds in flight, which DeepLabCut is might be better suited to. 
(I'm not sure I'm ready for MMPose either)

davidpagnon, 2025-05-20:
Okay, I'm running out of ideas, you can send me your folder if you are okay with it :)

Heard-of-Checkerboard, 2025-05-20:
Sure thing David, I'll zip up everything I can and email it to you (the folders keep failing to upload here). Thank you so much for sticking with me on this! 

I'll also say that running `pose_model = Hand` didn't work, but it seems like every model that isn't `Body_with_feet` is also causing problems. The error is massively long so I won't include it, but I think it stems from the system failing to download the model properly, which is another mystery.

davidpagnon, 2025-05-20:
Hi, I just had a look, your problem comes from your Config.toml file. The strings were not closed in several instances:
- For example, in the definition of your CUSTOM model, you had `  name = "mcarp_2` (mcarp_2 is never closed), and same in other instances
- At some point, you spelled CUCTOM instead of CUSTOM
- Python is very specific about its use of spaces and tabs. There are cases where you had tabs instead of 4 spaces, and line 285, you had 5 spaces instead of 4.
- You cannot run pose estimation with Pose2Sim, unless it is an onnx/zip file deployed from MMPose (RTMPose, RTMO, etc). So you need to keep "mode" commented

I don't know which text editor you are using, but I would recommend you to make sure you have syntactic coloring for toml files, as unmatched quotes would become more visible.

Here is the corrected table:

```toml
[pose.CUSTOM]
name = "wrist"
id = 0
  [[pose.CUSTOM.children]]
  name = "mcarp_1"
  id = 1
     [[pose.CUSTOM.children.children]]
     name = "pphal_1"
     id = 2
        [[pose.CUSTOM.children.children.children]]
        name = "dphal_1"
        id = 3
  [[pose.CUSTOM.children]]
  name = "mcarp_2"
  id = 4
    [[pose.CUSTOM.children.children]]
    name = "pphal_2"
    id = 5
        [[pose.CUSTOM.children.children.children]]
        name = "mphal_2"
        id = 6
           [[pose.CUSTOM.children.children.children.children]]
           name = "dphal_2"
           id = 7
  [[pose.CUSTOM.children]]
  name = "mcarp_3"
  id = 8
     [[pose.CUSTOM.children.children]]
     name = "pphal_3"
     id = 9
        [[pose.CUSTOM.children.children.children]]
        name = "mphal_3"
        id = 10
           [[pose.CUSTOM.children.children.children.children]]
           name = "dpahl_3"
           id = 11
  [[pose.CUSTOM.children]]
  name = "mcarp_4"
  id = 12
     [[pose.CUSTOM.children.children]]
     name = "pphal_4"
     id = 13
        [[pose.CUSTOM.children.children.children]]
        name = "mphal_4"
        id = 14
           [[pose.CUSTOM.children.children.children.children]]
           name = "dpahl_4"
           id = 15
  [[pose.CUSTOM.children]]
  name = "mcarp_5"
  id = 16
     [[pose.CUSTOM.children.children]]
     name = "pphal_5"
     id = 17
        [[pose.CUSTOM.children.children.children]]
        name = "mphal_5"
        id = 18
           [[pose.CUSTOM.children.children.children.children]]
           name = "dphal_5"
           id = 19
```

---

Then there was another problem: the likelihoods estimated by DeepLabCut were generally pretty low, so I set `likelihood_threshold_triangulation = 0`. The triangulation results were still not good, so I checked the calibration data and the pose estimation results. I think the model you trained is not very good: when I run `json_display_without_img -j <JSON_FOLDER>, the points are mostly red and do not move like a hand. 

https://github.com/user-attachments/assets/6fcad129-5702-4485-be02-9948d15c2e30

So I tried to run with the native `pose_mode = 'Hand'` and `mode = 'lightweight'`. Results are not great because the hand model is not good, but at least we've got something. 
https://github.com/user-attachments/assets/cb09f60a-b3ac-46c5-8b74-0dddce9b0c57

Be aware that I had to edit a few lines in triangulation.py, and that I have not pushed the change yet. If you want to try it, do `pip show pose2sim` to find where your file is stored, and replace the 3 lines after L708 with those:
```python
    try:
        keypoints_names_swapped = ['L'+keypoint_name[1:] if keypoint_name.startswith('R') else 'R'+keypoint_name[1:] if keypoint_name.startswith('L') else keypoint_name for keypoint_name in keypoints_names]
        keypoints_names_swapped = [keypoint_name_swapped.replace('right', 'left') if keypoint_name_swapped.startswith('right') else keypoint_name_swapped.replace('left', 'right') if keypoint_name_swapped.startswith('left') else keypoint_name_swapped for keypoint_name_swapped in keypoints_names_swapped]
        keypoints_idx_swapped = [keypoints_names.index(keypoint_name_swapped) for keypoint_name_swapped in keypoints_names_swapped] # find index of new keypoint_name
    except:
        keypoints_names_swapped = keypoints_names
        keypoints_idx_swapped = keypoints_idx
        logging.warning('No left/right swap was performed.')
```

However, I would recommend you downloading the Hand model from DeepLabCut, as it seems to be better. See this to find it:
https://anipose.readthedocs.io/en/latest/tutorial.html#generating-2d-labels:~:text=We%20have%20prepared,Anipose%20project%20folder.

Heard-of-Checkerboard, 2025-05-20:
Thanks for the update David, looks like I got myself into a fairly embarrassing mess on my end. My lack of experience is clearly showing here. 

I use Notepad++ to edit, so I'll look into adjusting the settings to make sure I don't run into similar problems again. 

I knew the DLC model I gave you needed a lot of work, so I appologise for the poor data. I'm in a bit of a time-crunch and I honestly just need to prove everything can fit together as a proof-of-concept, so I was telling myself refinement comes later. Clearly that was a mistake. Since we last spoke I've been training a new model, which I'm hoping will provide better likelihood thresholds. 

The hand model you've linked to is based on Tensorflow and I wasn't sure how well that would work in a Pytorch-based system like yours, which is why I avoided using it. I'll try again with that model to see if I'm worried for no reason. 

Also, thank you for the adjustments in the triangulation file. I appreciate you going the extra mile for me. 

I'll try out these adjustments and get back to you tomorrow. Thanks again!

davidpagnon, 2025-05-20:
No worries, it takes time! 
Notepad++ does have a toml extension, but I would use VScode or pycharm, which will offer much more along the way.

Well, if you train your model with DeepLabCut, you need to run pose estimation within DeepLabCut, and then to convert the results to the Pose2sim format. So it does not matter whether it runs on tensorflow or anything else. Unless, now that I think about it, unless you convert your tensorflow model to an onnx file, which you could run from within Pose2Sim. That may be the best option, actually!

I cannot do it just like that, but an LLM may be happy to help you. If you do, please tell me how it goes!

(And by the way, how did you train your model if not with deeplabcut?)

davidpagnon, 2025-05-20:
I just asked Claude.ai, I haven't read or checked but if you have a chance to try it out, feel free to tell me about it and then I might implement it in Pose2Sim (not sure when, though :)

# Converting a DeepLabCut Model to ONNX

Yes, you can convert a DeepLabCut model to ONNX format. DeepLabCut models are typically based on TensorFlow, so the conversion process is similar to what we discussed earlier, but with some specific considerations for handling the particular architecture and output format.

Here's a step-by-step guide to convert a DeepLabCut model to ONNX and use it with the output format you need:

## Step 1: Export the DeepLabCut Model

First, you need to export the DeepLabCut model. If you haven't already exported it, you can do so with:

```python
import deeplabcut
path_config_file = 'path/to/your/config.yaml'
trained_network_path = deeplabcut.export_model(path_config_file, iteration=None, shuffle=1, trainingsetindex=0)
```

## Step 2: Convert to ONNX

```python
import tensorflow as tf
import tf2onnx
import onnx
import os
import numpy as np

# Load your exported DeepLabCut model
model_path = "path/to/exported/deeplabcut/model"  # The exported TF model directory
dlc_model = tf.saved_model.load(model_path)

# Get the serving function
predict_fn = dlc_model.signatures["serving_default"]

# Determine input shape from the model
input_details = list(predict_fn.structured_input_signature[1].values())[0]
input_shape = input_details.shape.as_list()

# Create a concrete serving function with explicit shapes
# DeepLabCut typically expects RGB images with shape [batch, height, width, 3]
batch_size = 1  # Usually we process one image at a time
height, width = input_shape[1], input_shape[2]  # Get dimensions from the model
channels = 3  # RGB

# Define input signature for the model
input_signature = [tf.TensorSpec(shape=(batch_size, height, width, channels), 
                                dtype=tf.float32, name="input_image")]

# Convert to ONNX
onnx_path = os.path.join(os.path.dirname(model_path), "dlc_model.onnx")

try:
    # Convert using the concrete function and save
    model_proto, _ = tf2onnx.convert.from_function(
        predict_fn,
        input_signature=input_signature,
        opset=13,
        output_path=onnx_path
    )
    print(f"Conversion successful. ONNX model saved to {onnx_path}")
except Exception as e:
    print(f"Direct conversion failed: {e}")
    
    # Alternative: Use command line for conversion
    print("Trying command line conversion...")
    cmd = f"python -m tf2onnx.convert --saved-model {model_path} --output {onnx_path} --opset 13"
    import subprocess
    result = subprocess.run(cmd, shell=True, check=True)
    print(f"Command line conversion successful. ONNX model saved to {onnx_path}")
```
[... 135 lines cut ...]
for i, (keypoint, score) in enumerate(zip(keypoints, scores)):
    print(f"Keypoint {i}: Position {keypoint.tolist()}, Confidence {score.item():.4f}")
```

## Additional Tips for DeepLabCut Models

1. **Model-specific adjustments**: The exact output processing might vary based on your specific DeepLabCut model version and configuration. You might need to adjust the post-processing logic.

2. **Refinement with locref**: Some DeepLabCut models use location refinement (locref) to improve keypoint accuracy. If your model includes this, you'll need to include code to apply these refinements.

3. **Alternative with direct Python API**: DeepLabCut also offers a more direct route through its Python API:

```python
import deeplabcut

# This returns the ONNX model path
onnx_path = deeplabcut.export_model(path_config_file, trainIndex=None, shuffle=1, 
                                    trainingsetindex=0, onnx=True, make_tar=False)

# Then you can use the ONNX model as shown above
```

4. **Handling different image sizes**: The wrapper includes code to resize input images, but you might want to refine this based on your specific needs.

Would you like me to elaborate on any particular aspect of this process?

Heard-of-Checkerboard, 2025-05-20:
I'll have a look into those editors, thanks for the recommendation!

I did train my model with DeepLabCut, the latest update is based on Pytorch. They currently support both Tensorflow and Pytorch engines, I just happened to use Pytorch out of convenience. Maybe that will make things easier? 

When I have time I'll give this ONNX conversion script suggestion a try, or if I switch over to a Tensorflow build of DLC.

davidpagnon, 2025-05-20:
Oh okay, I did not realize they added the support for pytorch! It should not make a difference, either for you or for Pose2Sim--unless you want to run your model from within Pose2Sim.

Heard-of-Checkerboard, 2025-05-20:
That's good to know! I really wasn't sure if there was an issue.

Thanks again David, talk to you tomorrow.

Heard-of-Checkerboard, 2025-05-21:
Hi David, your suggestions worked and we are past the previous error! But now I have a new one in its place.
I'm working with the linked DLC model above, I've converted its h5 files to JSON as in the documentation, and set the Config correctly this time. However, now I'm getting an Attribute error (the one below is when I attempt to triangulate), and it looks like this:

`AttributeError                                                                            Traceback  (most recent call last)
Cell In[2], line 1
----> 1 Pose2Sim.triangulation()

File ~\minicondaenvs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:384, in triangulation(config)
      383 def triangulation(config=None):
--> 384        pipeline = Pose2SimPipeline(config)
      385        pipeline.triangulation()

File ~\minicondaenvs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:169, in Pose2SimPipeline.__init__(self, config)
      167 except:
      168        self.session_dir = os.path.realpath(os.getcwd())
--> 169 use_custom_logging = self.config_dicts[0].get('logging').get('use_custom_logging')
      170 if not use_custom_logging:
      171       setup_logging(self.session_dir)

AttirbuteError: 'NoneType' object has no attribute 'get'`

Should I be filling the argument with something and, if so, what specifically?

Heard-of-Checkerboard, 2025-05-21:
Hi David, I'm going through things now and there's good news and bad news.
Good news is the Config edits worked! I'll certainly be looking for those errors more closely. I'm currently attempting to run the "Hand" model as you did to see if I can get the kinematics from an OSIM I've built.

Bad news is the triangulation doesn't seem to be working, with or without the edits you put together. I keep getting this:
`ValueError:  'LWrist' is not in list`
Turning `handle_LR_swap=` to `true` or `false` doesn't seem to be having an effect. 
Did I miss something?

Thank you so much for your hard work yesterday! I will admit that I've also been trying to use the linked DLC model from earlier and I ran into an attribute error, but I'll cross that bridge later. I think I found a native Pytorch-ONNX model converter from Pytorch themselves, but one thing at a time.

davidpagnon, 2025-05-21:
That's where you need to replace the 3 lines L708, I'll push it in the next release but I'll wait for a few more edits to come before doing so.
You're welcome!

See this commit:
https://github.com/perfanalytics/pose2sim/commit/f91f6c4e59b14b85bbdb97651cdc009347961a5b

Heard-of-Checkerboard, 2025-05-21:
Y'know, I could have sworn it didn't work the first time, but now it is! Maybe I typed it in incorrectly again and VScode didn't flag it (i.e. I didn't set VScode to flag it properly.)

I suppose my only other question right now is how to associate my own OpenSim model for kinematics? Is that something I can have Pose2Sim do itself or do I need to do that manually in OpenSim?

davidpagnon, 2025-05-22:
I think there must still be a problem in your configuration file, which stops being read before the logging line.
Can you do 

```python
import toml
toml.load(r"<your_file.toml>")
```

and see if all the configuration parameters are printed?

Heard-of-Checkerboard, 2025-05-22:
Hi David,
Sorry for the late reply, I've been trying to put together my other OpenSim files. When I use that code, the last line of the printout is `'logging': {'use_custom_logging': False}}`, which is the last line after what looks to me to be the rest of the Config parameters from the config file. Should I be seeing more lines after the logging line?

Heard-of-Checkerboard, 2025-05-23:
I looked over the files again and I need to apologize again. I forgot to revert the Config file to the earlier version that caused the errors so it obviously didn't reproduce the errors a second time. With the DLC model back into the config file, running

`import toml
toml.load(r"<your_file.toml>")`

produces the following:

`{'project': {'multiperson': False,
 'participant height': 'auto',
 'participant_mass': 70.0,
 'frame_rate': 'auto',
 'frame_range': 'auto',
 'exclude_from_batch': []},
'pose': {'vid_extension': 'mov', 'pose_model': 'CUSTOM'}}`

I suspect this is closer to the error you were expecting. I'm sorry for misleading you earlier, my mind really was somewhere else. Does this at least get us closer to a solution?

davidpagnon, 2025-05-24:
That's how you learn!
It shows that it does not read the config file past "pose_model", so there must be an error right after it, maybe a quote you forgot to close or a misplaced comment.

If you cannot find it, please attach your file and I'll see how I can fix it!

Heard-of-Checkerboard, 2025-05-26:
Okay, that worked, the file now reads in full!

But now I'm getting a new ValueError, and I'm not sure what to change. It's like it's looking for a parameter in my JSON files that isn't there. 

The full error is
```
`ValueError                                               Traceback (most recent call last)
Cell In[9], line1
----> 1 Pose2Sim.triangulation()

File ~\minoconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:385 in triangulation(config)
    383 def traingulation(config=None):
    384       pipeline = Pose2SimPipeline(config)
--> 385      pipeline.triangulation()

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:243 in Pose2SimPipeline.triangulation(self)
    241 self._log_step_header("Triangulation of 2D points", confing_dict)
    242 start = time.time()
--> 243 triangulate_all(config_dict)
    244 elapsed = time.time() - start
    245 logging.info(f'\nTriangulation took {time.strftime("%Hh%Mm%Ss", time.gmtime(elapsed))},\n")

File ~\miniconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:736, in triangulate_all(config_dict)
    733 json_files_names = [sort_stringlist_by_last_number(js) for js in json_files_names]
    735 #frame range selection
--> 736 f_range = [[0,min([len(j) for j in json_files_names])] if frame_range in ('all', 'auto', []) else frame_range][0]
    737 frame_nb = f_range[1] - f_range[0]
    739 #Check that camera number is consistent between calibration file and pose folders

ValueError: min() arg is an empty sequence`
```
I've tried playing around with the camera names as suggested by line `739` but it doesn't seem to have an effect. Did I need to define my camera timeline and forgot?

davidpagnon, 2025-05-26:
It probably means that one of the folders in your "pose" directory is empty, or that their names do not correpsond to the names of the cameras in the calibration file.

davidpagnon, 2025-05-26:
Congrats! Glad it finally worked :)
I'm closing it for now, feel free to reopen it or to create a new issue if needed.

closed 2025-05-26 by davidpagnon