perfanalytics/pose2sim #191: Error during Demonstration Part-1: No model specified

AirChord, 2025-07-01:
Dear all.

I am first experiencing with Pose2Sim I am following the installation and demonstration present on the github repository, and when I try to run Demonstration Part-1 can't make it work, when it reaches the Pose2Sim.kinematics() it returns with the following error:

File "/home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.10/site-packages/Pose2Sim/kinematics.py", line 489, in perform_scaling
    opensim.ScaleTool(scaling_path_temp).run()
  File "/home/ubuntu/miniconda3/envs/Pose2Sim/lib/python3.10/site-packages/opensim/tools.py", line 2130, in run
    return _tools.ScaleTool_run(self)
RuntimeError: std::exception in 'bool OpenSim::ScaleTool::run() const': ScaleTool: No model specified.

Can you help or suggest something.
I am running this on ubuntu 20.04 and on conda environment, I am just testing to check if I can use pose2sim on my project, so I'm just using CPU.

Thanks in advance.

AirChord, 2025-07-02:
Hi David, Thanks for the reply, I did a source install, and my version is Version: 0.10.25, I uninstalled the source installation and installed using pip install pose2sim. i still have the same problem.
Below is a more detailed clip of the error.

Scaling...
[info] Updating Model file from 40000 to latest format...
[info] Loaded model Pose2Sim_WithContactsAndMuscles from file /home/ubuntu/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/OpenSim_Setup/Model_Pose2Sim_muscles_flex.osim
[info] Processing subject Pose2Sim_scaled...
[info] Step 1: Loading generic model
[info] Loaded model Pose2Sim_WithContactsAndMuscles from file /home/ubuntu/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/Demo_SinglePerson/kinematics/Demo_SinglePerson_1-47_filt_butterworth_LSTM.osim
[error] Error: Dimensions the WrapEllipsoid radii cannot be less than or equal to 0.
[error] Unable to load the generic model or marker set file.
[error] ScaleTool: No model specified.
Error during scaling for pose-3d/Demo_SinglePerson_1-47_filt_butterworth_LSTM.trc: std::exception in 'bool OpenSim::ScaleTool::run() const': ScaleTool: No model specified.\.
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[1], line 9
      7 Pose2Sim.filtering()
      8 Pose2Sim.markerAugmentation()
----> 9 Pose2Sim.kinematics()

File ~/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/Pose2Sim.py:397, in kinematics(config)
    395 def kinematics(config=None):
    396     pipeline = Pose2SimPipeline(config)
--> 397     pipeline.kinematics()

File ~/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/Pose2Sim.py:268, in Pose2SimPipeline.kinematics(self)
    266 self._log_step_header("OpenSim scaling and inverse kinematics", config_dict)
    267 start = time.time()
--> 268 kinematics_all(config_dict)
    269 elapsed = time.time() - start
    270 logging.info(f'\nOpenSim scaling and inverse kinematics took {time.strftime("%Hh%Mm%Ss", time.gmtime(elapsed))}.\n')

File ~/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/kinematics.py:678, in kinematics_all(config_dict)
    675 logging.info(f"Processing TRC file: {trc_file.resolve()}")
    677 logging.info("\nScaling...")
--> 678 perform_scaling(trc_file, pose_model, kinematics_dir, osim_setup_dir, use_simple_model, right_left_symmetry=right_left_symmetry, subject_height=subject_height[p], subject_mass=subject_mass[p], 
    679                 remove_scaling_setup=remove_scaling_setup, fastest_frames_to_remove_percent=fastest_frames_to_remove_percent, large_hip_knee_angles=large_hip_knee_angles, trimmed_extrema_percent=trimmed_extrema_percent,close_to_zero_speed_m=close_to_zero_speed)
    680 logging.info(f"\tDone. OpenSim logs saved to {opensim_logs_file.resolve()}.")
    681 logging.info(f"\tScaled model saved to {(kinematics_dir / (trc_file.stem + '_scaled.osim')).resolve()}")

File ~/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/kinematics.py:480, in perform_scaling(trc_file, pose_model, kinematics_dir, osim_setup_dir, use_simple_model, right_left_symmetry, subject_height, subject_mass, remove_scaling_setup, fastest_frames_to_remove_percent, close_to_zero_speed_m, large_hip_knee_angles, trimmed_extrema_percent)
    477 scaling_tree.write(scaling_path_temp, pretty_print=True, xml_declaration=True, encoding='utf-8')
    479 # Run scaling
--> 480 opensim.ScaleTool(scaling_path_temp).run()
    482 # Remove scaling setup
    483 if remove_scaling_setup:

File ~/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/opensim/tools.py:2130, in ScaleTool.run(self)
   2121 def run(self):
   2122     r"""
   2123      Run the scale tool. This first runs the ModelScaler, then runs the
   2124     MarkerPlacer. This is the method called by the command line `scale`
   (...)
   2127     :return: whether or not the scale procedure was successful.
   2128     """
-> 2130     return _tools.ScaleTool_run(self)

RuntimeError: std::exception in 'bool OpenSim::ScaleTool::run() const': ScaleTool: No model specified.

AirChord, 2025-07-03:
Hi David,
Of course i can send you that.

davidpagnon, 2025-07-03:
Thanks for sending me your directory.

It is very strange, because I cannot see any problem on my end, including with your files.
Can you maybe check this file: `~/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/kinematics.py`,
add this line 446:
```python
        print(unscaled_model_path)
        print(scaled_model_path)
```
And send me the files you'll find at these locations?

By the way, I just found something fishy when the synchronization GUI is on in the single-person Demo. I just fixed it and pushed the change, so you could reinstall pose2sim with `pip install pose2sim -U`. It should not have anything to do with your issue, though.

AirChord, 2025-07-06:
Thanks for your fast repply
I checked the file: ~/miniconda3/envs/Pose2Sim2/lib/python3.10/site-packages/Pose2Sim/kinematics.py, and added your suggestion to the line 446, and the expected files are there.
I sent the files to you the same way as before.
I reinstalled with the new version and as you expected nothing was solved, the error continues.

AirChord, 2025-07-07:
Hi David,
It worked, i was able to run the example generate the .mot file.
When visualizing in the Opensim GUI, I think on moved a little weird but that another issue, maybe due the way i ran the example.
Thanks for the help

closed 2025-07-07 by AirChord

davidpagnon, 2025-07-09:
Yes, I think there might be a problem with your synchronization, which in turn messed up your triangulation. For now, you can delete your "pose-sync" folder and rerun it without `Pose2Sim.synchronization()`, you should have better results.