perfanalytics/pose2sim #194: Problem with synchronization()

Dylan1202, 2025-07-31:
Hi, thank you for your great work on Pose2Sim! I have a quick question about the synchronization() function.

According to the documentation, I understand that a window should appear to let me choose which person to synchronize. However, when I run the function, no window pops up—it just proceeds directly with the calculation.

[image]

I’ve confirmed that my video contains multiple people, and I’ve already set multi_person: true in the config file. Is there anything else I need to set to make the selection window appear? Or is there a way to force it to show up?

Thanks in advance for your help!

davidpagnon, 2025-08-01:
Hum, this is strange 🤔

Do you use Mac? Did you try basic matplotlib plotting in the pose2sim environment?

Dylan1202, 2025-08-01:
> Hum, this is strange 🤔
> 
> Do you use Mac? Did you try basic matplotlib plotting in the pose2sim environment?

I’m actually using Windows, not macOS. I tried plotting with matplotlib in the Pose2Sim environment and it works fine—images display without any issue. I’ll also try running the same files on another computer to check if it might be something specific to my local configuration.